StateSmith/StateSmith #21: feature - PlantUML import

adamfk, 2022-09-20:
> Great project, thank you for sharing! I will definitely try this out for my next state machine.
> How about adding a [PlantUML](https://plantuml.com/de/state-diagram) import option? I believe puml is a pretty common language to draw diagrams in code.

[reddit suggestion](https://www.reddit.com/r/embedded/comments/xj6b4k/comment/ip6u1uf/?utm_source=share&utm_medium=web2x&context=3)

This is a great idea! We should be able to support a text front end without too much difficulty. We just need to supply the StateSmith compiler with a list of `DiagramNode`'s and `DiagramEdge`'s. 

[YedParser](https://github.com/StateSmith/StateSmith/blob/main/src/StateSmith/Input/Yed/YedParser.cs#L17-L18)
[DiagramNode](https://github.com/StateSmith/StateSmith/blob/9b36dbc301b8275ea1be7795853d4aa2a987e7a3/src/StateSmith/Input/DiagramNode.cs#L17-L20)
[DiagramEdge](https://github.com/StateSmith/StateSmith/blob/9b36dbc301b8275ea1be7795853d4aa2a987e7a3/src/StateSmith/Input/DiagramEdge.cs#L17-L20)

If someone would like to help with this, let me know and I'll help get you started.

jnz86, 2022-09-21:
Agreed. This should be a priority.

I've been through some HSM work lately, and the only thing holding me back from giving StateSmith a serious eval is I don't want another diagramming format. 

I personally and professionally use PlantUML in documentation and markdowns. Both are superior to an XML output for git diffs. I can not see adding a specific, closed source, proprietary, albeit free yEd into my required toolchain. Especially not a program that I need to leave VSCode to work with (those days are pretty much over).

I am normally put off by code gen, but I think you have something decent here. All I can provide is perspective that PlantUML (or some form of GraphViz) is a make or break feature.

adamfk, 2022-09-21:
@jnz86 Would you be able to share a sample plantuml state diagram so that I can use it for testing against? It will help me focus my parsing efforts (do I need to support diagram styling right away?).

I 1000% agree about XML being terrible for git diffs. I definitely want to improve this.

FYI - I tried looking for a plantuml state diagram ANTLR4 grammar last night, but didn't find anything. If you come across anything, please let me know! Otherwise, I'll get started on an ANTLR4 grammar.

adamfk, 2022-09-26:
@jnz86 PlantUML input support is ready in the dev branch. Details: https://github.com/StateSmith/StateSmith/blob/dev/docs/plantuml-input.md

As noted in the documentation above, you just need to feed StateSmith 0.5.6 with a PlantUML file.

Let me know if you run into any problems.

commit 3ee08239 (2022-09-25): plantuml - documentation, examples, skinparam fix

#21
closed 2022-09-26 by adamfk