StateSmith/StateSmith #57: Example integrating with Zephyr RTOS

beriberikix, 2022-10-14:
Hello! Love the idea behind StateSmith. I was wondering if you had any examples showing how to integrate StateSmith into an RTOS like [Zephyr](https://www.zephyrproject.org/). Either by integrating with its [State Machines Framework](https://docs.zephyrproject.org/latest/services/smf/index.html) or show an alternative.

Cheers.

adamfk, 2022-10-14:
Hi :)

I currently use StateSmith with FreeRTOS. It should work with Zephyr as well because it has no dependencies on any OS or anything (it can run baremetal). Are you running into issues?

I do plan to properly support multiple different code generation strategies in the future. This would make it easy for you to write a bit of code so that StateSmith would output code to work with Zephyr's state machine framework.

I don't currently have a stable API for that yet, but if you want to explore it anyway, take a look at https://github.com/StateSmith/StateSmith/blob/main/src/StateSmith/compiler/SpecificVertexValidator.cs

You would likely need to create your own vertex visitor to visit the different vertices in the state machine graph.

amigomcu, 2022-10-18:
hi ,
zephyr os which have include fsm function,

https://docs.zephyrproject.org/3.1.0/services/smf/index.html

> Hello! Love the idea behind StateSmith. I was wondering if you had any examples showing how to integrate StateSmith into an RTOS like [Zephyr](https://www.zephyrproject.org/). Either by integrating [...]

adamfk, 2023-02-14:
Hi @beriberikix, are you still interested in having StateSmith output Zephyr SMF code? If so, I can create a template project to help get you started. We now have initial support for custom code generation.

adamfk, 2023-03-12:
Doesn't look like there is any interest in this. Closing for now. We can reopen if someone would like to work on it.

closed 2023-03-12 by adamfk