StateSmith/StateSmith #69: Python real time GUI front end

wmxsteel, 2022-12-25:
You can use something like Dearpygui. It’s a intermediate gui and it already has a drag and drop diagramming feature

Great work on state smith, I definately want to help out. I am a battery management systems controls engineer and I write embedded code on firmware for a living. 

[https://www.reddit.com/r/DearPyGui/comments/wwsvv9/showcase_app_wedx_made_with_dear_pygui/](url)

adamfk, 2022-12-26:
The imnodes editor looks really nice
https://github.com/Nelarius/imnodes

I'm not sure it would be good for nesting states, but it looks really helpful for data flow or filtering.

This demo screen looks like it has a kind of nesting though:
https://github.com/hoffstadt/DearPyGui/blob/assets/readme/nodes1.png

I'll document some of the features I'm looking for in a GUI soon. The main one is wanting to be able to collapse states/groups.

adamfk, 2023-02-02:
@wmxsteel are you interested in pursuing Dearpygui now that we have draw.io integration working?

adamfk, 2023-02-16:
I'm going to close this for now, but we can reopen if there is interest in this again.

closed 2023-02-16 by adamfk