StateSmith/StateSmith #75: orthogonal states

adamfk, 2022-12-31:
Not too hard to implement, but a lot of work to fully test.

0Grit, 2023-10-12:
@adamfk A little rambling off the top of my head since I've been thinking a lot about UML orthogonal states lately and just came across StateSmith.

Orthogonal regions seem to open up for the possibility to create states/machines which are portable and independent but still affectable by events happening in another state/machine.

Example: operation of a bank of LEDs, and and the actions that happen based on the state of a button may differ depending on the 'mode' state of an application.

Orthogonal states could enable rapid changes like making an orthogonal state for an encoder and using it in place of the button 'state' to control the applications 'mode' state.

Sounds super powerful... but in order to retain true independence between orthogonal states some translation of events would need to be coordinated by the parent state. (essentially the  application itself)

Assuming I want to keep independence between orthogonal states, I've not fully thought through how to handle this translation within a C based framework.

I also noticed that it looks like StateSmith evaluates events from the 'bottom up' versus  starting with the outermost state, which could pose a problem.

Again just thinking out loud as I continue to digest these more 'advanced' state machine topics and frameworks implementing them.

adamfk, 2023-10-13:
@0Grit an earlier version of state smith (not released) supported orthogonal states. They are super helpful in larger state machine driven applications (less so in small low level state machines). Bottom up event propagation still works, but yes, it is more tricky to implement.

I definitely intend to support orthogonal states, but it likely won't happen for a year or more. I'm currently super focused on another embedded project: https://github.com/fin-language/fin/