StateSmith/StateSmith #76: feature - easier time tracking

adamfk, 2022-12-31:
support syntax like 
* `after 500ms`
* `after 500 ms`
* `after some_var ms` ? `some_var` could be an expansion or variable.
* `[$ms_in_state > 400]` ?
* `[$time_in_state > 400 ms]` ?
    * how to implement? support post fix expansions? `ms(string preText)`?

For this to work StateSmith will need:
1. a way to get the current time. Provided via an expansion.
2. a way to know the scaling of time provided by the above. Does the above return milliseconds? Seconds? Microseconds?

emmby, 2024-05-23:
being able to support periodic behaviors with something like `every 10 seconds` would also be cool

emmby, 2024-05-23:
Auto-off:
```mermaid
stateDiagram

off
on

off  --> on: BUTTON
on --> off: after 30 seconds
```

Blinky:
```mermaid
stateDiagram

off
on

off --> on: every 1 second
on --> off: after 200ms
```

As I write this, I'm a little unclear on the semantics of `every`. I know that the previous tool I used supported it, but in this example it's not clear how Blinky would be different if I just replaced `every` with `after`.