StateSmith/StateSmith #100: draw.io - replace more special space characters

adamfk, 2023-02-11:
replace more special space characters listed here https://www.compart.com/en/unicode/category/Zs

adamfk, 2023-02-11:
Let me know if you are interested in tackling this issue and I can help guide you. It is a good first issue.

FyrBridd, 2023-02-15:
Hi, I would be interested in taking this as my first issue! I assume all that is needed is to add the additional characters from the list.

adamfk, 2023-02-15:
Thanks!!! Yep. Just update the MxCellSanitizer and the MxCellParserTests files where the ticket is linked.

In case you haven't already seen it: https://github.com/StateSmith/StateSmith/wiki/Building-Source-and-Running-Tests

Let me know if you run into any issues.

adamfk, 2023-02-15:
Can you also update the change log and add your name? Kinda like this:

```
## WIP
### Added
- draw.io - convert more special characters to regular space characters (@FyrBridd).
  - https://github.com/StateSmith/StateSmith/issues/100
```

closed 2023-02-20 by adamfk