StateSmith/StateSmith #101: feature - generate verilog for FPGA/ASIC

adamfk, 2023-02-13:


PlantinumX, 2023-02-15:
Hi I would be interested.

adamfk, 2023-02-15:
Awesome! ❤️

How about we start small? Can you draw up a small state machine (maybe 3 states) and the verilog code that we should generate for that diagram? It doesn't need to perfect and we can change it as we work, but it will help give us an initial target.

PlantinumX, 2023-02-23:
Hi, I could deliver a first draft in the following days/weeks.

adamfk, 2023-02-23:
Sounds great! No rush. If you ever want to discuss some ideas, we can use discord: https://discord.com/channels/1056394875278462996/1056394875278462999

I'm always happy to talk about state machines :)

PlantinumX, 2023-02-25:
Hi the discord link is broken or something . It does not work for me... :)

adamfk, 2023-02-25:
Hmm... I'm not very experienced with Discord. Can you try this link instead? https://discord.com/invite/rNxNGQXWsU

adamfk, 2023-04-12:
Hi @PlantinumX. How's the FPGA work coming along? Need any help? Or should we put this on the back burner for a while?

PlantinumX, 2023-04-14:
Hi @adamfk , currently I do not have time to work :/. Pls put it on the back burner for a while :)

adamfk, 2023-04-15:
Sounds good @PlantinumX :)