StateSmith/StateSmith #105: Won't/can't unit test the generated state_id_to_string() or a ROOT_exit() functions

rahmant3, 2023-02-17:
Hi, this is a great project! =) 

I like unit testing the generated source code to verify the behavior but I don't want to unit test these functions. Since I don't unit test them, my code coverage metrics show them as not covered.

- ...state_id_to_string() -- I don't personally use this function at all so it would be nice if this was excluded from the code coverage metrics
- ROOT_exit() -- This function is unreachable as my state machine doesn't typically exit (they just run forever).

The following comments could be added before/after these functions to exclude them from code coverage metrics:

```
// LCOV_EXCLUDE_START
// GCOV_EXCLUDE_START
// GCOVR_EXCLUDE_START
... function goes here ...
// LCOV_EXCLUDE_STOP
// GCOV_EXCLUDE_STOP
// GCOVR_EXCLUDE_STOP
```

This should exclude the function from being picked by lcov, gcov, and gcovr.

adamfk, 2023-02-20:
Great suggestion!

adamfk, 2023-02-24:
Completed as part of #114. See https://github.com/StateSmith/StateSmith-examples/tree/main/user-post-process

Video tutorial will be uploaded soon-ish to https://www.youtube.com/@statesmith

closed 2023-02-24 by adamfk

adamfk, 2023-02-25:
Video links:
* [Removing state id to string function](https://www.youtube.com/watch?v=UYL6Qzoyxwo&t=614s)
* [Surrounding a function with code coverage annotations ](https://www.youtube.com/watch?v=UYL6Qzoyxwo&t=810s)