StateSmith/StateSmith #109: add fsm_event_id_to_string() API

iamjuing, 2023-02-21:
For logging porposes, it would be nice to be able to log events by name.

adamfk, 2023-02-22:
I like this idea, but I will probably wait to implement until I add more code generation customizations. Some platforms (like AVR8, small Arduinos) need to handle strings specially to [avoid putting them in RAM](https://www.arduino.cc/reference/en/language/variables/utilities/progmem/).

In the mean time, here's something that I do to map the name of the events:

```c
// https://stackoverflow.com/a/4415646/7331858
#define COUNT_OF(x) ((sizeof(x) / sizeof(0 [x])) / ((size_t)(!(sizeof(x) % sizeof(0 [x])))))

const char * const event_names[] = {
    [Spec2Sm_EventId_DO] = "DO",
    [Spec2Sm_EventId_EV1] = "EV1",
    [Spec2Sm_EventId_EV2] = "EV2",
    [Spec2Sm_EventId_EV3] = "EV3",
    [Spec2Sm_EventId_EV4] = "EV4",
    [Spec2Sm_EventId_EV5] = "EV5",
    [Spec2Sm_EventId_EV6] = "EV6",
    [Spec2Sm_EventId_EV7] = "EV7",
    [Spec2Sm_EventId_EV8] = "EV8",
    [Spec2Sm_EventId_EV9] = "EV9",
    [Spec2Sm_EventId_EV10] = "EV10",
    [Spec2Sm_EventId_EVBACK] = "evBack",
    [Spec2Sm_EventId_EVCLOSE] = "evClose",
    [Spec2Sm_EventId_EVOPEN] = "evOpen",
    [Spec2Sm_EventId_EVSTEP] = "evStep",
};
static_assert(COUNT_OF(event_names) == Spec2Sm_EventIdCount, "above mapping outdated");
static_assert(Spec2Sm_EventIdCount == 15, "above mapping outdated");
```

iamjuing, 2023-02-22:
Understand. The problem is that storing data in flash is not only platform specific, it is also compiler specific.
"__flash" in IAR vs. "PROGMEM" in GCC vs. "const" for CPUs with unified address space.
Moreover, placement in the definition also varies.

iamjuing, 2023-02-24:
One possible solution of staying independent from storage qualifiers could be to generate only a literal with names that the client code stores as per its needs, and provide a mapping from EventId enum to an index to that literal.

adamfk, 2023-02-24:
Completed as part of #114. See https://github.com/StateSmith/StateSmith-examples/tree/main/user-post-process

Video tutorial will be uploaded soon-ish to https://www.youtube.com/@statesmith

closed 2023-02-24 by adamfk

adamfk, 2023-02-25:
Video link: [custom generating event to id function](https://www.youtube.com/watch?v=UYL6Qzoyxwo&t=951s)

Custom user code that generates the event id to string function is actually pretty small:

```c#
    // c# user code generation script
    public string FunctionCode()
    {
        var sm = smProvider.GetStateMachine();
        var sb = new StringBuilder();
        var output = new OutputFile(codeStyle, sb);

        output.Append(FunctionSignatureCode());
        output.StartCodeBlock();
        {
            output.Append("switch (id)");
            output.StartCodeBlock();
            {
                foreach (var eventName in sm.GetEventListCopy())
                {
                    output.AppendLine($"case {mangler.SmEventEnumValue(eventName)}: return \"{eventName}\";");
                }
            }
            output.FinishCodeBlock();
            output.AppendLine("return \"?\";");
        }
        output.FinishCodeBlock();
        return output.ToString();
    }

    string FunctionSignatureCode()
    {
        var output = "// Converts an event id to a string. Thread safe.\n";
        output += $"const char* {mangler.SmName}_event_id_to_string(const enum {mangler.SmEventEnum} id)";
        return output;
    }
```

```c
// Generated .c code

// Converts an event id to a string. Thread safe.
const char* MySm_event_id_to_string(const enum MySm_EventId id)
{
    switch (id)
    {
        case MySm_EventId_DO: return "do";
        case MySm_EventId_EV2: return "ev2";
        case MySm_EventId_MYEV1: return "myev1";
    }
    return "?";
}
```