StateSmith/StateSmith #111: draw.io: hard dependency on element style

iamjuing, 2023-02-23:
It seems there is hard dependency on the plugin for the VScode draw.io (3rd party) plugin.
My goal was to use the draw.io desktop version, however, I found out I had to use specific styling of the elements, especially for text events in the container elements. There is a dependency like gradientColor, strokeColor etc.

Is such a dependency really necessary? It seems the desktop version draw.io can not be extended with the plugin, right?
Any chance to support standard UML elements directly provided by draw.io (perhaps More Shapes -> Software ->UML 2.5)?
Thanks.

adamfk, 2023-02-23:
Right now, you can use draw.io without any special shapes as long as the node labels match what StateSmith is expecting. This is similar to how the yEd diagrams are drawn. You can actually download the below svg and rename it back to `Tutorial1Sm.drawio.svg` to allow draw.io to edit it. It works with StateSmith.
[image]

You can also copy the StateSmith shapes from this [online template](https://viewer.diagrams.net/index.html?tags=%7B%7D&highlight=0000ff&edit=_blank&layers=1&nav=1&title=StateSmith%20Template.drawio#Uhttps%3A%2F%2Fdrive.google.com%2Fuc%3Fid%3D1yLK0es2Jf9LB6FRSt3loK6Pkxe8vlxo0%26export%3Ddownload) if you want.

More info here if you haven't seen it yet: https://github.com/StateSmith/StateSmith/wiki/Getting-started-using-draw.io-with-StateSmith

## UML 2.5 shapes
I wasn't aware of these! Thanks for pointing them out. Is there a good shape for composite states? Something like `ON_GROUP` in the image below?

[image]

Would you be able to share a sample diagram that you would like to use and then we can figure out a way of parsing it with StateSmith?

adamfk, 2023-02-23:
We need to keep at least some of those [styling restrictions](https://github.com/StateSmith/StateSmith/wiki/Getting-started-using-draw.io-with-StateSmith#restrictions-on-styling-) so that we can parse the diagram reliably.

In the draw.io xml file, text nodes and shape nodes can be basically the same thing. StateSmith needs some way of knowing that the text belongs to the parent and isn't a nested state.

The below styles ensure that the text node doesn't look like a state node:
```
fillColor=none;
gradientColor=none;
strokeColor=none;
```
We could relax the validation though. Instead of making sure that all the above are all set to `none`, we could also accept if there was no style for them.

We could potentially relax the below restrictions as well.
* `resizable=0;`
* `movable=0;`
* `deletable=0;`
* `rotatable=0;`

The main thing I wanted to prevent was diagrams with the behavior code in strange places:

[image]

I think initially, we could add a flag in [DrawIoSettings.cs](https://github.com/StateSmith/StateSmith/blob/main/src/StateSmith/Runner/DrawIoSettings.cs) like 
```c#
public bool relaxTextEventHandlerValidation = false; // opt-in for now
```

Thoughts?

iamjuing, 2023-02-24:
I totally understand you need to be able to distinguish among the elements.
I have no further thoughts. I was just wondering if it was possible at all to eliminate the need for the plugin by using the already provided UML 2.5 elements.

adamfk, 2023-02-24:
I like the idea of it. I'm thinking we could start with the basics:

- [ ] recognize regular circle shape as initial state without needing the label `$initial_state`
- [ ] recognize rhombus/diamond shape as choice state without needing the label `$choice`
- [ ] recognize circle with "H" label as shallow history state
- [ ] implement `relaxTextEventHandlerValidation` as discussed above

Any interest in helping me?

iamjuing, 2023-02-24:
Unfortunately, I don't have any spare time to help with coding nowadays. Sorry :(
I investigated the UML 2.5 elements in drawio though... some hints...

**shape=ellipse**: either "junction" or "initial" ... distinguish programatically (junction must always have an incoming transition, initial shall not have an incoming transition)

**ellipse with value=H**: shallow history

**ellipse with value=H* **: deep history

**rhombus**: always a choice point

**shape=mxgraph.sysml.actFinal**: final transition

**shape=mxgraph.sysml.simpleState**: only suported state state (can be turned into composite state by enabling Container property)

regions: ?

adamfk, 2023-04-18:
This page has some good info we could look into supporting: https://www.diagrams.net/blog/uml-state-diagrams