StateSmith/StateSmith #112: replace "#pragma once" with include guard

iamjuing, 2023-02-23:
"#pragma once" is non-standard compiler extensions. Although supported by many compilers, I would still recommend to avoid it in favour of the regular include guard generated from the FSM name.

adamfk, 2023-02-23:
I was wondering about this myself recently. Pragma once is great for code/file headers that might be copy pasted around, but that probably isn't going to happen with the generated FSM code.

Do you have a suggested default rendering setting? If state machine name is `MySm` include guard would be 

```c
#ifndef MYSM_H
#define MYSM_H

#endif /* MYSM_H */
```

We can also add a customization to override the default rendering.

iamjuing, 2023-02-23:
Looks good to me.

commit 504b2222 (2024-08-29): lib - C/C++ - Add support for standard `#ifdef` include guards

Closes #112
closed 2024-08-29 by adamfk