StateSmith/StateSmith #118: expansions in diagrams

adamfk, 2023-02-26:
# Implementation options
We could embed a javascript engine and have a lot of confidence in security, but then we add another syntax for expansions which I'm not a fan of. Also adds more complexity and size to StateSmith.

I'm leaning towards using Roslyn.

In the diagram:

[image]

If you paste from vscode and enable html label in draw.io, you could see this instead:

[image]

One day we could have our syntax highlighting live.

leading with `public` is optional and can be omitted

[image]

We could also potentially allow the leading `string` to be omitted (but this is more difficult and requires new custom parsing which I would rather avoid):

[image]

## Transformed Into
```c#
using StateSmith.Input.Expansions; // for UserExpansionScriptBase
public class DiagramExpansion_DiagramIdHere : UserExpansionScriptBase
{
    public string boom3() => "light_boom3()";

    public string timer1_started_at_ms => AutoVarName();
    public string option_value         => AutoVarName();
    
    public string set_option_class(string class_name) => $"{option_value} = PlayerClass_{class_name}";
    public string time_ms => "PortApi_get_time_ms()";
    public string reset_timer1() => $"{timer1_started_at_ms} = {time_ms}";
}
```

Then parsed with Roslyn to ensure no security concerns.

## Security
Right now, expansions can execute any C# code. This is great if you want to be able to do something specialized. If you need that, you'll have to run StateSmith from a .cs or .csx file.

```c#
using StateSmith.Input.Expansions;
public class DiagramExpansion_DiagramIdHere : UserExpansionScriptBase
{
    public string option_value         => System.SomethingDangerous("","");
}
```

### Fail on any SimpleMemberAccessExpression

[image]
[image]

### Need to prevent any implicit usings.
Need to look into.

### Need to prevent object creation
Can fail on `NewKeyword` token.

[image]

[image]

[image]

[image]

### Anything else to consider?
Definitely not future proof as C# language can evolve. I wonder if there is a way to ask Roslyn to parse as a certain C# version? Would be safer to only allow certain syntax or use js engine.

PopeAndrewJ, 2023-08-20:
I wonder if something like 
[DotNetIsolator ](https://github.com/SteveSandersonMS/DotNetIsolator)  will come out of  experimental any time soon..

adamfk, 2023-08-20:
@PopeAndrewJ that is a cool project! Thanks for sharing it. It would definitely help solve this problem.

How concerned do you think we should be about security and potentially malicious code inside an expansion? Should StateSmith just prompt the user if it is going to execute expansion code that they haven't already approved?

adamfk, 2024-07-22:
Closing for now. I think it's more complicated and powerful than we need in diagrams. We will support simple expansions in diagrams, but if you need full power expansions, then use .csx scripts.

closed 2024-07-22 by adamfk