StateSmith/StateSmith #126: Generate idiomatic C++

adamfk, 2023-03-10:
We can generate C++ code now with the help of [customizing](https://github.com/StateSmith/StateSmith/blob/main/examples/ButtonSm1Cpp/CodeGen/ButtonSm1Cpp.cs#L26-L28) the Generic Intermediate Language (GIL) to C99 transpiler, but it isn't very convenient.

The generated code is also C99 code that happens to be C++ compliant.

It would be nice to generate proper C++ code with classes.

Some thought needs to go into this though.
* do we generate single header only file? or .hcpp and .h?
* what version of c++?

It would be helpful if someone could translate a simple state machine into the desired C++ code.

You could start from the blank example [.c file](https://github.com/StateSmith/StateSmith/blob/main/examples/Tutorial1-blank/src/Tutorial1Sm.c) and [.h file](https://github.com/StateSmith/StateSmith/blob/main/examples/Tutorial1-blank/src/Tutorial1Sm.h).

iamjuing, 2023-03-10:
Just a quick, blind try how it could look like (pretty natural I guess)...

[Tutorial1Sm.cpp.txt](https://github.com/StateSmith/StateSmith/files/10943063/Tutorial1Sm.cpp.txt)
[Tutorial1Sm.h.txt](https://github.com/StateSmith/StateSmith/files/10943064/Tutorial1Sm.h.txt)

adamfk, 2023-03-10:
this looks good. I really like the idea of having the state machine optionally extend a user class. This could help eliminate the need for most StateSmith expansions which is good. It also gets us one step closer to not needing a .csx file at all.

Thoughts on using C++ enum classes instead?

Instead of 
```cpp
enum StateId
{
	StateId_ROOT = 0,
	StateId_OFF = 1,
	StateId_ON1 = 2,
};
```

```cpp
enum class StateId
{
	ROOT = 0,
	OFF = 1,
	ON1 = 2,
};
```

iamjuing, 2023-03-13:
"StateId_" prefix makes no sense of course.
When it comes to the "enum class", it is C++11. I would recommend to stay C++03 compatible.
It should be fine if it is encapsulated in the class, however, what is really needed is to think of supporting orthogonal regions (maybe in form of standalone FSMs on different DrawIO pages?), which need to share the same "enum EventId".
One way would be enclose all in a namespace not to accidentally collide with signals of unrelated FSMs.

adamfk, 2023-03-14:
> It should be fine if it is encapsulated in the class

Thanks for pointing that out. I thought that regular C++ enums acted like C enums.

We will eventually get [orthogonal regions](https://github.com/StateSmith/StateSmith/issues/75), but it might take some time. It's a lot of work that I'm not too excited to do right now.

Sharing a common event enumeration is a very interesting idea though. Would help with users manually implementing orthogonal regions.

bloodgain, 2023-09-21:
> ...When it comes to the "enum class", it is C++11. I would recommend to stay C++03 compatible.

At this point, all major compilers are well beyond C++11 compatibility, and all platforms that ship with a non-C++11 compiler are well beyond their support dates. The point of generating idiomatic C++ should be to support current C++ developers, who have moved on to modern C++. I've worked on a legacy code that predates the adoption of the STL, and even those projects have now mostly adopted modern compilers since Red Hat 6 reached end-of-life in 2020.

A good cutoff at this point is C++14. Full GCC support here just reached 10 years old, and should cover nearly anyone looking at this library to generate significant amounts of code.

The rare developer still stuck using decades-old compilers could still generate C99 state machines using the project. Cross-compatible libraries were the norm when these compilers were in common use, so this will be familiar ground for that small subset of users.

adamfk, 2023-09-23:
Good points @bloodgain. However, even if StateSmith generates C++03 code for boiler plate and transition logic, users can still write C++11 or newer code in their diagrams. I think the only feature I wanted for StateSmith from C++11 was enum classes, but @iamjuing has a good solution for this.

commit bec32f4c (2024-10-04): changelog update for cpp. Closes #126
closed 2024-10-04 by adamfk