StateSmith/StateSmith #127: Documentation for setting up a new StateSmith project from scratch

sdaglish, 2023-03-10:
It would be useful to have some documentation on how to setup a minimum StateSmith project from scratch, rather than copying and then modifying one of the example projects.

For example, if I was adding this to a current project, I'd want to set everything up in a project subfolder (lets say statesmitchfsm), add the plugin, add an empty svg file, add the csx file. Then only the minimum number of files necessary for the plugin to generate the SM header and source files. 

I'd then include these generated files in my parent project without the requirement for any other example source files.

adamfk, 2023-04-24:
Hi @sdaglish. Can you check out this new page? https://github.com/StateSmith/StateSmith/wiki/draw.io:-new-project

closed 2023-04-30 by adamfk

adamfk, 2024-05-07:
@sdaglish I just released a new tool to make setting up a project take seconds instead of minutes: https://github.com/StateSmith/StateSmith/blob/main/src/StateSmith.Cli/README.md