StateSmith/StateSmith #136: Support "[else]" guard for internal transitions

iamjuing, 2023-03-13:
Shall support this scenario w/o a need for using guard check ordering:

[image]

As per discussion: https://github.com/StateSmith/StateSmith/discussions/134

adamfk, 2023-03-14:
@iamjuing can you download [this example project](https://github.com/StateSmith/StateSmith-examples/tree/main/proposal-else-guard) and try out the implementation? 

Alternatively, you can just copy [this code to your project](https://github.com/StateSmith/StateSmith-examples/blob/main/proposal-else-guard/src/MySm.csx#L13-L37). Might need to copy [some of the usings too](https://github.com/StateSmith/StateSmith-examples/blob/main/proposal-else-guard/src/MySm.csx#L13-L37).

I was thinking specifying an order like `1. EVENT [else]` should be illegal.

iamjuing, 2023-03-14:
I used the code snippet in my CSX, tried various combinations and it works well.
Yes, any ordering applied to `EVENT [else]` should be definitely considered illegal.

iamjuing, 2023-04-03:
Found an issue ... with this case:
[image]

Generated output:
[image]

Problem:
`if (cnt==1)` block does not have a return statement and thus falls to `else` block.

iamjuing, 2023-04-03:
Could you please make this a priority into an official release?

adamfk, 2023-04-03:
> Found an issue ... with
> Problem:
> if (cnt==1) block does not have a return statement and thus falls to else block.

That behavior is expected with the current StateSmith event handling. STATE_1 has 3 behaviors:
* EVENT [cnt == 1]
* EVENT [cnt == 2] TransitionTo(STATE_3)
* EVENT [else]  TransitionTo(STATE_2)

`EVENT [cnt == 1]` will prevent an ancestor state from handling `EVENT`, but still allows STATE_1's other `EVENT` behaviors to run. Only a transition will prevent any other behaviors from being checked.

[image]

This is how you can achieve the desired effect:

[image]

Although this is clearer:

[image]

(the above svg diagram is editable with draw.io)

Make sense?

adamfk, 2023-04-03:
> Could you please make this a priority into an official release?

Yep. I'll try to get it out in the next few days.

adamfk, 2023-04-03:
See plantuml diagram links in tests: https://github.com/StateSmith/StateSmith/blob/bc10435e4756eadf94f85358373d084aa200107e/src/StateSmithTest/SmGraph/ElseGuardProcessorTests.cs

iamjuing, 2023-04-04:
The solution you suggest is not equivalent as it exits and re-enters the state which is not desirable.
Where does this statement come from: "_Only a transition will prevent any other behaviors from being checked._" ?

The thing is that `EVENT [cnt==1]` is a transition. It is an internal transition (with guard) as per the UML standard.
Therefore , StateSmith shall threat it as a transition and the first `if (cnt==1)` block shall contain the `return` statement to prevent side-effect by actually activating the `EVENT [else]` external transition.

adamfk, 2023-04-04:
> The solution you suggest is not equivalent as it exits and re-enters the state which is not desirable.

True. I added more examples and my reasoning here: https://github.com/StateSmith/StateSmith/discussions/158 Let's continue the discussion there. These discussions are super helpful to me because they help broaden my perspective so that we can make StateSmith as useful as possible.

I've also added this wiki page to help collect differences with UML:
https://github.com/StateSmith/StateSmith/wiki/UML-Differences

commit bc10435e (2023-04-03): Support `else` guard. Closes #136