StateSmith/StateSmith #137: Add option to auto prefix states so each is Fully Qualified

adamfk, 2023-03-14:
See discussion: https://github.com/StateSmith/StateSmith/discussions/125

I don't think this will be the default, but we can make it an option people can turn on. Could be very helpful for very large designs.

adamfk, 2023-03-14:
Here's a quick potential implementation: https://github.com/StateSmith/StateSmith-examples/tree/main/proposal-fully-qualified-state-ids

I didn't spend much time on it. If anyone is interested in this idea, can they please help test it?

iamjuing, 2023-03-14:
I would still prefer this (from the discussion):
_As a trade-off, the solution could be just to generate the state name in for "parentState_childState" **only for the duplicates**, and only up to the different parent name._

So, unless there are duplicate states (which I guess is rare), everything would look like it is now.
Is there anything you find could be a problem? I can imagine that logic is much more difficult to code.

adamfk, 2023-03-14:
> I would still prefer this (from the discussion): _As a trade-off, the solution could be just to generate the state name in for "parentState_childState" **only for the duplicates**, and only up to the different parent name._

Me too.
 
> So, unless there are duplicate states (which I guess is rare), everything would look like it is now. Is there anything you find could be a problem? I can imagine that logic is much more difficult to code.

Here's a separate ticket for shortest fully qualified names: https://github.com/StateSmith/StateSmith/issues/138

Once we determine the strategy for naming, the coding will be easy. I just want to make sure that our naming strategy considers every possibility.

> It could even be the default behavior.

Agreed. Once we are sure that the implementation is thoroughly tested.

adamfk, 2023-03-29:
Closing in favor of https://github.com/StateSmith/StateSmith/issues/138

closed 2023-03-29 by adamfk