StateSmith/StateSmith #167: consider adding `$stop_event`

adamfk, 2023-04-14:
Would help achieve UML like behavior where a non-transition behavior (UML internal transition) will prevent a transition.

See discussion https://github.com/StateSmith/StateSmith/discussions/158

[image]

`$stop_event` could be implemented to prevent ANY more state behaviors from running for that event dispatch.

iamjuing, 2023-04-14:
This has to be available at the diagram level otherwise it would be pretty tedious to draw the (UML like) diagram. 
I would not complicate it to be "at transition" level. It would be also easier to implement and test it. 

So, if the `terminal_transitions_enabled` global config flag is:
- `true`, all the transitions are considered terminal and `return` statement is simply generated in each block
- `false` would be the default value retaining the current StateSmith behavior (`return` only in external transitions' blocks)

This brings us to your previous idea of `$statemachine(uml): MySm` where the `(uml)` could just enable that flag (and perhaps others in the future). There could be also more modifiers in the future, that would just set/clear particular configuration flags used by the generator.

What do you think?

adamfk, 2023-04-15:
I agree. I've added ticket #170 for `$statemachine(uml): MySm`. While I like the name `terminal_transitions_enabled`, I think I would prefer it be in the diagram to hopefully help prevent user confusion `$statemachine(uml): MySm`. Hopefully we start to see more StateSmith diagrams in the wild in the future :)

Sorry I wasn't very clear in the ticket description, but I was thinking `$stop_event` would be used for StateSmith semantics for those (hopefully) few places you need it. Not intended to be used everywhere to achieve UML compliance. I agree, that would be very tedious.

I'm thinking we add `$stop_event` support first, test the implementation out, and then if we like it, we can advance on to #170.

Thoughts?

iamjuing, 2023-04-17:
Whatever makes sense to you if you find that useful for StateSmith behavior as well.
I'm just struggling a bit with the name `$stop_event` as it has nothing to do with the event itself but rather with the processing behavior (it does not mean that other transitions with the same event are not evaluated - depending on order). Maybe some alternatives related to processing:
`$end`, `$stop`, `$final`, `$consume`, `$terminate`, `$end_processing`, `$stop_processing`

**Related, but different topic....**

By the way, I am not following other languages you add to the StateSmith however, could `EVENT / $whatever` collide with syntax of any of them? I would really avoid putting "keywords" to the action area.
Better to define some unique and unified way of writing StateSmith keywords, similar e.g. to well known (to software designers) UML stereotypes, in angle brackets.

`$STATEMACHINE<UML>`
`EVENT [guard] <final> / action()`
`EVENT [guard] <consume> / action()`
`<consume> EVENT [guard]  / action()`
`<1> EVENT [guard]  / action()`
`<2> EVENT [guard]  / action()`

... are there any other already supported?

iamjuing, 2023-04-17:
There could be also an opposite keyword as well, if you decide to bring in a global (default behavior) switch, which I hope for.
No need to refer to UML at all with `$STATEMACHINE<UML>` actually.

If so, there could be something like this supported:

`<transition> EVENT [guard]  / action()` to put "return" to the action block if the global switch is disabled 
`<behavior> EVENT [guard]  / action()` to remove "return" from the action block if the global switch is enabled

adamfk, 2023-04-18:
> I'm just struggling a bit with the name $stop_event as it has nothing to do with the event itself but rather with the processing behavior.

Excellent point! I'm still thinking with the features we have right now, but `$stop_event` is a terrible name (for its intended purpose) when orthogonal states are added.

UML stereotypes are a good idea. Opened https://github.com/StateSmith/StateSmith/issues/176 for it.

I've been meaning to open a ticket regarding `consume_event` being too likely to clash with user code. Good call regarding other languages. Opened ticket https://github.com/StateSmith/StateSmith/issues/175