StateSmith/StateSmith #176: consider stereotypes

adamfk, 2023-04-18:
Excellent suggestion from Juing: https://github.com/StateSmith/StateSmith/issues/167#issuecomment-1510796896

UML stereotypes are a good idea. Nice syntax. Could be very useful for users wanting to add custom extensions: find all states with a stereotype and add custom behavior. Cleaner than using [`$mod`](https://github.com/StateSmith/StateSmith-examples/tree/main/modding-logging).

I actually like the syntax better for ordering behaviors as well.
```
<2> EVENT1 / stuff( );
2. EVENT1 / { stuff( ); /* somewhat user error prone */ }
```

The problem with `2. EVENT1 / stuff( );` is that users (myself included) sometimes forget the period and write `2 EVENT1 / stuff( );` which is invalid syntax. More likely to happen if someone uses a decimal as well. `2.0 EVENT1 / stuff( );` is invalid. Would need to be `2.0. EVENT1 / stuff( );`. Perhaps we could just drop the requirement of needing the dot after the number.

iamjuing, 2023-04-19:
Just a note ... multiple stereotypes shall be possible I guess, e.g.:
`<behavior> <1> EVENT [guard]  / action()`

adamfk, 2023-04-27:
@iamjuing I'd like to start with stereotypes for states first. How does the below syntax look? I'm thinking we support either single or double angle brackets. The stereotype would have to come before the state name.

[image]

Note: to be able to change the swimlane title divider height, you have to edit the shape's style to make it rotatable or manually set `startSize` to something like 45.
[image]

iamjuing, 2023-04-27:
Looks good to me. The double angle brackets is actually what the UML standard uses. I would go with that.
Having more stereotypes:

[image]

adamfk, 2023-04-27:
Good call on multiples. I think I'll leave the parsing grammar very permissive while we experiment. You'll be able to put anything between the angle brackets: `<1>`, `<<menu(2, 4)>>`. We can lock it down later after we gather feedback and experience using it.

iamjuing, 2023-04-27:
Ok.

Let's define to what elements stereotypes can be applied:

- $STATEMACHINE
- state behaviours (except entry/exit)
- external transitions
- simple states
- composite states

Placement: always first

Let's defined also the initial list of StateSmith reserved ones:

state behavior: <transition>, <N>
external transition: <N>

N ... code generation order

adamfk, 2023-04-28:
Sounds good to me. I'll implement the standard StateSmith reserved stereotypes as a transformation step so that people can easily disable that if they want to.

[image]