StateSmith/StateSmith #179: feature - sub state machine

adamfk, 2023-04-18:
This might take a little while to while to implement, but would be very useful for re-using behaviors. Great for robotics and automation.

Before we go implement the feature, I want to test out how to do it DIY first. I'll be adding a new example here: https://github.com/StateSmith/StateSmith/issues/178

It would also be very helpful for large teams and large designs so that they could work on separate files concurrently without tons of git merge conflicts. One team works on `MainSm`, and another works on `SubSm`. `MainSm` references `SubSm`.

## Implementation notes
The Balanced1 algorithm RAM function pointers might allow an efficient implementation with minimal cost.

## Question 1: Do we allow recursion?
Can a sub state machine reference itself directly or indirectly? Initially, I think probably not.

## Question 2: Should the sub state machine stand on its own as well?
Could be helpful for testing. Or does the sub state machine need to always be embedded inside a proper state machine first?

## Syntax suggestions?

iamjuing, 2023-04-19:
Some thoughts/comments ...

1. a state machine shall not allow to reference itself
2. not sure what you mean by "on its own". There is nothing like "sub state machine", let's take UML that refers to "submachine state" which is one of the state types. It just has to be part of a state machine just like any other state.

**Some implementation considerations**
1) overlapping event enums ... each state machines has its own enumerator
This will not be effortless ... the designer must provide an event translation table. The parent state machine would call a translation function whenever it is in a submachine state, defined externally like e.g. `child_event_enum translate_event(parent_event_enum e_id)` or so. No effort shall be required in the diagram itself.

2) event and state string name generation
Could be in a form of: "parent submachine state name / child state machine state name" provided in these ways:
- dynamically: This would however, require the client code to supply a long enough buffer and involve string copying.
- statically: StateSmith to generate single constant string for each possible combination: This could grow exponentially, imagine a submachine can reference other submachine etc. (it's RAM vs. speed tradeof as always)
- none: generator config (e.g. enabled for debug/development images only)

3) a state machine can contain multiple "submachine states" referencing the same state machine

adamfk, 2023-04-19:
@iamjuing if you have time, do you think you could make a small toy example for us to discuss?

adamfk, 2023-04-20:
@iamjuing had an interesting idea:
> Just an idea...
Submachine implementation could go also in other, simpler direction:
Understanding submachine as a C macro ... just to replace recurring parts of one diagram. There could be a $SUBMACHINE element in the same document, basically a composite state defined out of the main diagram.

I had wondered about that too. It is pretty easy to support. My fear is that it could quickly lead to very large generated code. It would be like copy pasting all the draw.io nodes. It would work OK if the submachine is small, but most of the submachines that I want to use are more like 20+ states.

If we want to implement the macro/copy paste way, maybe it could have different syntax? `$STATIC_SUBSM : SomeName`. Not sure how I feel about that.

iamjuing, 2023-05-16:
Drawing more complex diagrams, I've realized that the "macro" way of implementing submachines (or whatever we call it) would allow not only to reuse recurring diagram parts, but also to hide complex composite states thus, making diagram at different levels of abstraction. If there are more complex composite states in the diagram, it's getting started overcrowded and even unreadable, especially for review.

"_If we want to implement the macro/copy paste way, maybe it could have different syntax? $STATIC_SUBSM : SomeName. Not sure, but it feels like users should be very aware of the costs of using it._"
I'm pretty fine not to mix it with submachines. We could simply call it `$MACRO` which fits well I guess and is a well known term. 

I still don't feel comfortable however, with supporting submachine states that would allow running completely different and independent state machines from other state machine. I do not feel any benefit in that, as the designer would have to interconnect everything manually (different state machines react to different event etc.), just like in case of having more communicating state machines. I don't see StateSmith to be helpful here. This might be also the reason why other modeling tools and generators out there only understand submachines as the diagram decomposition feature.

"_My fear is that it could quickly lead to very large generated code._"
Well, large functionality requires large diagrams which always require more memory. This is a matter of implementation though. It does not necessarily need to be inserted to every single spot it is referenced in. However, it could be a good and safe first step opened for future optimisation.

Anyway, feel free to separate this out to its own issue as a feature independent of submachines.

adamfk, 2023-05-18:
> to hide complex composite states thus, making diagram at different levels of abstraction. If there are more complex composite states in the diagram, it's getting started overcrowded and even unreadable, especially for review.

Have you tried collapsing a composite state? You can see an example of this here: https://github.com/StateSmith/StateSmith-drawio-plugin/wiki/How-to-Use#visually-collapse-a-state

I like the `$MACRO` name. Nice suggestion! Opened a new ticket for it here: https://github.com/StateSmith/StateSmith/issues/201

iamjuing, 2023-05-18:
I have. The problem with collapsing states though is that it's messing up the layout, while there still need to be enough space to be able to fully open them. Moreover, the composite states can grow over time. I find collapsing basically unusable here.

adamfk, 2023-05-18:
@iamjuing if you have time, I would be really interested to understand the issue you are having with collapsing states.

Started a discussion here: https://github.com/StateSmith/StateSmith/discussions/202