StateSmith/StateSmith #190: $NOTES element sometimes fails code generation

iamjuing, 2023-05-02:
**Works well:**
[image]

**Fails:**
[image]

**Output:**
```
VertexValidationException: Notes vertices cannot have any behaviors
    Vertex
    Path: StateMachine{test_fsm}.$Notes
    Diagram Id: fDqNpV3gBqAaSbMREs_j-58
    Children count: 0
    Behaviors count: 1
    Incoming transitions count: 0
```

**Version:** 9.13-alpha

adamfk, 2023-05-02:
By design, StateSmith doesn't allow notes vertices to have incoming or outgoing edges. I see that the validation is missing the case where the $notes vertex exists outside of the state machine.

What should we do here?

1. allow notes vertices to have edges?
2. improve validation to prevent any edges on notes?

I like what you've drawn and I can see that being useful so I'm in favor of allowing $notes vertices to have edges. 

# Q1. Should we require any particular styling? Dashed? No arrow head? Or free for all?
I think I'm OK with free for all. Requiring a particular style is a fair bit more work. Style info currently isn't brought into state machine vertices.

[image]

# Q2. Should we allow edges between notes?
I think I'm ok with this.

[image]

# Q3. Should we allow edges from a regular state machine vertex to a $notes vertex?
I think this one should not be allowed. What would it even mean?

[image]

# Q4. Allow unterminated/floating edges from $notes?
Could be useful if we want to put a note for a transition. This one may be more work.

[image]

iamjuing, 2023-05-03:
Having connectors on edges is very common. There is no point to forbid that.
Need to allow note vertices to have any kind of connector. This is up to the designer just like having a comment in a code anywhere. StateSmith shall simply ignore any connector going to/from $NOTES element.

By the way, it would be very handy to extend note capability by this element from the UML library (`shape=note2`) so we do not need to have the redundant $NOTES tag:
[image]

JGBMichalski, 2023-05-03:
I think Q1 (free-for-all), Q2, and Q4 are good options. Not that I'll use them much (if at all), but I can see how that would be helpful.

adamfk, 2023-05-03:
Added a new ticket for supporting draw.io notes shapes: https://github.com/StateSmith/StateSmith/issues/192

commit 82ea6148 (2023-05-19): Relaxed `$notes` validation. You can now have notes point to states and vice versa.

Closes #190