StateSmith/StateSmith #192: draw.io - support note shapes

adamfk, 2023-05-03:
Good suggestion from: https://github.com/StateSmith/StateSmith/issues/190#issuecomment-1532501709

Might as well support all of the draw.io note shapes :)

[image]

This would be a nice alternative to the standard `$NOTES` shapes. Both will be supported.

[image]

iamjuing, 2023-05-04:
To be general, would `shape=.*note.*` regex work safely?

adamfk, 2023-05-05:
I'm a bit hesitant to use a regex here because I can't be sure. Right now we only have 3 notes shapes that I'm aware of so I think I'll just stick with testing against those three for now. If we find users wanting to add custom notes shapes, we could also support a custom `STATESMITH_NOTES_VERTEX=1;` style. Safe and then can be applied to whatever users want.

commit 559e5943 (2023-05-19): draw.io - support note shapes. Closes #192.
closed 2023-05-19 by adamfk