StateSmith/StateSmith #193: Add support for final state (default exit point)

iamjuing, 2023-05-04:
[image]

adamfk, 2023-05-05:
This will be really helpful for when we have sub state machines #179 and also orthogonal states https://github.com/StateSmith/StateSmith/issues/75.

I don't think we can use a final state as a default exit point though as a final state is not a pseudo state. Can you elaborate?

Here's a tiny example with an exit point. The exit point is a pseudo state and is transitioned through during an event dispatch.

[image]

One tricky thing to consider with final states, is that they rely on [completion events](https://docs.nomagic.com/display/CST190/Completion+Events+and+Transitions). We don't support automatic completion events because they tend to complicate implementations (queues, potential infinite loops...).

When we do support final states, what syntax should we use (see `???` below)? `via done`?

[image]

iamjuing, 2023-05-05:
This is useful for composite states as well.

**When we do support final states, what syntax should we use (see ??? below)? via done?**
... none. Just a connector w/o any label (allowing to have a guard though). This shall be perfectly sufficient and compliant with the UML.

Correct me if I'm wrong, but the internal representation can still be the same as the ExitPoint. I don't think we need to support completion events formally. StateSmith can treat the FinalState exactly the same as the ExitPoint. It would be however, associated with connectors only having a guard at most, thus labeled "[guard]". Just like ExitPoint "exit: X" is only associated with connectors labeled "via exit X".

adamfk, 2023-05-05:
Using the below example, assume `x` is 0, and we are in state `A`. When `EV1` is dispatched, the state machine will transition to `S1.FinalState` (which can't be a pseudo state) and have to stay there because the guard `[x > 20]` is false.

[image]

The UML spec has final states as a subclass of state and not a pseudo state which makes sense to me. This will make it easier to support orthogonal regions like below.

[image]

> Just a connector w/o any label (allowing to have a guard though).

Legacy StateSmith initially tried doing something like this, but it was too error prone for users without event queuing and completion events. See https://github.com/StateSmith/StateSmith/discussions/194

We could do something like this though:

[image]

iamjuing, 2023-05-10:
In the first example, StateSmith can detects the diagram is ill-formed as it can hang up. Support of `[else]` would be handy in this situation as there would be always at least one enabled transition which prevents the problem.

But, you are correct, the FinalState must really be a state in orthogonal regions as it serves as a synchronization barrier.

Well, it happens that a feature can be "error prone". That is pretty fine. Completion events are useful and have been defined in the UML for a reason. If the designer creates a loop in the diagram accidentally it's not different from creating an endless loop in code. Just like the compiler won't complain, the StateSmith shall not mind as it can be user's intention.

There is no dependency on event loop either. The completion event does not need to be put in a queue as it has a dispatch priority. Once generated, it is dispatched right away. Yes, a loop can be introduced however, intentionally as well.

UML `do` behavior is not supported. I have no problem with that. I don't find however, the "error prone" a reason not to fully support completion events (with an additional feature of StateSmith checking the diagram can get stuck in a FinalState).

Also, the decission here could be to support completion event generation only by reaching the FinalState. So, only a composite/statemachine state could have outgoing transitions with no trigger specified.
`entry` behaviors would just not generate completion events by default. Could be an optional feature though.

adamfk, 2023-05-10:
Is there a particular use case or example you can share? It might help my understanding. If I understand correctly, you would like final states to function like exit states (which are pseudo states). Is there any reason why you can't just use an exit state?

iamjuing, 2023-05-11:
No, not really. As you correcty pointed out, the FinalState must be a real state because of supporting orthogonal regions that depend on that.

I can certainly use the ExitPoint for composite state, however, I found the main advantage of using FinalState is that the environment of the composite state is not dependent on the ExitPoint label (especially if there is just one way out). The advantage might be even more significant in submachine states.

Let's say adding the FinalState generating completion events could be the first good step for future support of orthogonal regions. If orthogonal regions are not considered for StateSmith (which would be a valid decission), the FinalState could really be implemented like a nameless ExitPoint.

adamfk, 2023-05-12:
I think we can get most of the convenience by improving exit points with https://github.com/StateSmith/StateSmith/issues/195 and https://github.com/StateSmith/StateSmith/issues/196

adamfk, 2023-05-12:
Forgot to mention that orthogonal states are definitely something I plan to support (issue 75), but it will take a while to get there :)