StateSmith/StateSmith #197: Statesmith vs IAR Visual State Comparison

rahmant3, 2023-05-15:
I've been using Statesmith to create state machines, and I was recently evaluating IAR Visual State V11.1 to see if I should consider using it. I don't have any particular suggestions for improvements or issues, I'm just providing this feedback as is.

**_Statesmith won by a large margin! Here are the highlights._**

**Disclaimer:** I have been using Statesmith for a few months, whereas I've only used IAR Visual State the last week or so. Disclosing this to let you know there may be some unintended bias in my review here.

I spent a few days recreating a state machine I had created in Statesmith in IAR VS to evaluate it. I used both tools to generate C/C++ code for my state machine to see how it runs. I didn't have any issues with either tool when it came to running the generated code on my machine.

**What Statesmith does better:**
- Statesmith supports using user-defined symbols much better (they can be declared in a different header file)
- Statesmith supports writing inline C code
- Statesmith was more stable (IAR VS crashed several times through my use of it)
- Statesmith + Draw.io was much better experience than the built-in IAR Visual State Editor. 
  - IAR VS was very finicky and hard to get things to align properly compared to Statesmith. 
  - IAR VS is very restrictive in that it wants you to define all of the actions/events/variables within the tool itself, so it ends up being the case that you have to do a lot of clicking which I found tedious. 
  - IAR VS only lets you use basic types (integer, boolean, etc) when declaring actions, events, variables, etc, which makes it more work to interface with the generated code.
- I found Statesmith examples and documentation much easier to follow. IAR VS has a lot of documentation but it felt that I had to discover how to use the UI mainly on my own.
- Statesmith generated code was much easier to read than IAR VS. This is really helpful when I'm unit testing my generated state machine code and verify code coverage.
- Statesmith is free!

**What IAR VS did better:**
- IAR VS has a built-in simulator. My state machine was fairly simple so I didn't use this very much.
- IAR VS has a built-in debugger integration with IAR IDE. Again I didn't use this very much.
- IAR VS has a directory tree on the left of the editor that allows you to view your states in a tree-like structure (e.g. substates).
- IAR VS generates Statemachine documentation automatically containing diagrams and tables of the states, actions, transitions, etc. I imagine this is helpful to share with customers or store internally.
- IAR VS generated code is MISRA C/C++ compliant. This could be helpful to certify the state machine for functional safety.

Overall, Statesmith was a much better user experience. I can see what they are going for with IAR VS but it came across as a little unrefined/unfinished compared to Statesmith. Well done to the Statesmith team! It had lots of positives that I took for granted until I tried IAR VS.

iamjuing, 2023-05-17:
**Same experience here with the IAR VS:**
- insanely expensive
- non-public pricing
- obsolete UI (crashing)
- table-oriented state machine implementation - basically impossible to debug although it have advantages (less footprint I guess)
- although the "Classic Coder for C" option can be setup to generate readable code, I found out the output is actually buggy from my point of view - calling all guards of all outgoing transition on choice points to find the enabled transition, which can introduce hidden side-effects (hard to say to what extent this is by design though)

**Pros over StateSmith:**
- MISRA
- complies with the UML standard (very important one)
- feature-rich (StateSmith is however a very young tool to compare yet)
- generates C++, atlhough it unnecessarily uses polymorphism
- workspace approach (all product diagrams at the same place)

My overall feeling of IAR VS was that it is an obsolete tool for the cost.

adamfk, 2023-05-18:
Thanks for the feedback :)

closed 2023-05-18 by adamfk