StateSmith/StateSmith #200: feature - output file for easy git diff

adamfk, 2023-05-18:
This would be helpful for a few reasons:
* if a design uses `.drawio.svg` files, there isn't much git can do to help as the drawio data is base64 encoded into the svg file.
* the text file would only contain relevant info and should diff better than a regular `.drawio` xml file.
* if users are using advanced features and modding their diagrams, this would allow them to see exactly what is being fed into the code generation step

Concept:
```
DRINK
=================
Parent: ORDER
Diagram Id: 10
Behaviors:
    enter / x++;
    EV10 / y--;

DRINK.InitialState
=================
Parent: DRINK
Diagram Id: 11
Behaviors:
    TransitionTo(COLD)

COLD
=================
Parent: DRINK
Diagram Id: 12
Behaviors:
    EV1 / do_stuff( );
    EV2 / {
      multiline_action( );
    }
    EV3 TransitionTo(HOT)

HOT
=================
Parent: DRINK
Diagram Id: 13
Behaviors:
    EV1 / do_stuff( );
```

For larger designs, it would also be good to list behaviors "inherited" from ancestors:

```
COLD
=================
Parent: DRINK
Diagram Id: 12
Behaviors:
    EV1 / do_stuff( );
    EV2 / {
      multiline_action( );
    }
    EV3 TransitionTo(HOT)
Inherited behaviors from `DRINK`:
    EV10 / y--;
Inherited behaviors from `NORMAL_OPERATION`:
    EV_ERR TransitionTo(OUT_OF_ORDER)
```

iamjuing, 2023-05-23:
Not sure if this fits the topic ...
I was already thinking of comparing diagrams in a visual form for easy reviews. Found [svgdiff](https://www.svgdiff.dev/) with not success though.

adamfk, 2023-07-20:
I've been working on this a bit. Currently planning to output two sections in the same file:
1. before any transformations are run (TriggerMap, remove RenderConfig and notes). This is good for representing the input to StateSmith.
2. after all transformations are run. This is good for understanding what the code generation step sees.

Here's an example:

<details><summary>Details</summary>
<p>

[image]

##################################################
# Before transformations
##################################################

Vertex: <Notes>
=================
Type: NotesVertex
Diagram Id: 140
Notes:
	This is some stuff.
	
	Bold text.

Vertex: <RenderConfig>
=================
Type: RenderConfigVertex
Diagram Id: 141

Vertex: <Config>(AutoExpandedVars)
=================
Parent: <RenderConfig>
Type: ConfigOptionVertex
Diagram Id: 145
Option:
	int x;
	int y;

Vertex: NORMAL
=================
Parent: ROOT
Type: State
Diagram Id: 138
Behaviors:
	enter / { x = 0; }

	do / { x++; }

	exit / { x = 15; }

	[err] TransitionTo(ERROR)

	=========== from ancestor ROOT ===========

	ANY / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: ROOT
=================
Type: StateMachine
Diagram Id: 5
Initial State: ROOT.<InitialState>
Behaviors:
	ANY / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: ROOT.<InitialState>
=================
Parent: ROOT
Type: InitialState
Diagram Id: 8
Behaviors:
	TransitionTo(DELAY)

Vertex: DELAY
=================
Parent: NORMAL
Type: State
Diagram Id: 9
Behaviors:
	enter / { buzz(); }

	EV1 / { y = 33; }

	exit / { stop_buzz(); }

	[after_ms(100) 
	&& x >= 13] TransitionTo(RUNNING)

	=========== from ancestor NORMAL ===========

	do / { x++; }

	=========== from ancestor ROOT ===========

	ANY / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: ERROR
=================
Parent: ROOT
Type: State
Diagram Id: gO1ZRfetmGtumTaL4_i4-140
Behaviors:
	PRESS / { buzz(); }

	=========== from ancestor ROOT ===========

	ANY / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: RUNNING
=================
Parent: NORMAL
Type: State
Diagram Id: gO1ZRfetmGtumTaL4_i4-143
Initial State: RUNNING.<InitialState>
Behaviors:

	=========== from ancestor NORMAL ===========

	do / { x++; }

	=========== from ancestor ROOT ===========

	ANY / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: <RenderConfig>
=================
Parent: ROOT
Type: RenderConfigVertex
Diagram Id: gO1ZRfetmGtumTaL4_i4-146

Vertex: <Config>(TriggerMap)
=================
Parent: <RenderConfig>
Type: ConfigOptionVertex
Diagram Id: gO1ZRfetmGtumTaL4_i4-148
Option:
	// some comment
	ANY => * /* wildcard */

Vertex: PRE_HEAT
=================
Parent: RUNNING
Type: State
Diagram Id: gO1ZRfetmGtumTaL4_i4-150
Behaviors:
	enter / { set_timeout(5 min);
	x *= 23; }

	=========== from ancestor NORMAL ===========

	do / { x++; }

	=========== from ancestor ROOT ===========

	ANY / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: RUNNING.<InitialState>
=================
Parent: RUNNING
Type: InitialState
Diagram Id: gO1ZRfetmGtumTaL4_i4-152
Behaviors:
	TransitionTo(PRE_HEAT)

##################################################
# After transformations
##################################################

Vertex: NORMAL
=================
Parent: ROOT
Type: State
Diagram Id: 138
Behaviors:
	enter / { x = 0; }

	do / { x++; }

	exit / { x = 15; }

	do [err] TransitionTo(ERROR)

	=========== from ancestor ROOT ===========

	(INC_ERR, do, EV1, PRESS) / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: ROOT
=================
Type: StateMachine
Diagram Id: 5
Initial State: ROOT.<InitialState>
Behaviors:
	(INC_ERR, do, EV1, PRESS) / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: ROOT.<InitialState>
=================
Parent: ROOT
Type: InitialState
Diagram Id: 8
Behaviors:
	TransitionTo(DELAY)

Vertex: DELAY
=================
Parent: NORMAL
Type: State
Diagram Id: 9
Behaviors:
	enter / { buzz(); }

	EV1 / { y = 33; }

	exit / { stop_buzz(); }

	do [after_ms(100) 
	&& x >= 13] TransitionTo(RUNNING)

	=========== from ancestor NORMAL ===========

	do / { x++; }

	do [err] TransitionTo(ERROR)

	=========== from ancestor ROOT ===========

	(INC_ERR, do, EV1, PRESS) / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: ERROR
=================
Parent: ROOT
Type: State
Diagram Id: gO1ZRfetmGtumTaL4_i4-140
Behaviors:
	PRESS / { buzz(); }

	=========== from ancestor ROOT ===========

	(INC_ERR, do, EV1, PRESS) / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: RUNNING
=================
Parent: NORMAL
Type: State
Diagram Id: gO1ZRfetmGtumTaL4_i4-143
Initial State: RUNNING.<InitialState>
Behaviors:

	=========== from ancestor NORMAL ===========

	do / { x++; }

	do [err] TransitionTo(ERROR)

	=========== from ancestor ROOT ===========

	(INC_ERR, do, EV1, PRESS) / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: PRE_HEAT
=================
Parent: RUNNING
Type: State
Diagram Id: gO1ZRfetmGtumTaL4_i4-150
Behaviors:
	enter / { set_timeout(5 min);
	x *= 23; }

	=========== from ancestor NORMAL ===========

	do / { x++; }

	do [err] TransitionTo(ERROR)

	=========== from ancestor ROOT ===========

	(INC_ERR, do, EV1, PRESS) / { log("unhandled event"); }

	INC_ERR / { err++; }

Vertex: RUNNING.<InitialState>
=================
Parent: RUNNING
Type: InitialState
Diagram Id: gO1ZRfetmGtumTaL4_i4-152
Behaviors:
	TransitionTo(PRE_HEAT)

</p>
</details>

commit e6c8fdfe (2023-08-02): SmGraphDescriber - add settings to enable or disable different sections. closes #200
closed 2023-08-02 by adamfk