StateSmith/StateSmith #213: New feature to support PlantUML !include preprocessing

yongzhy, 2023-07-21:
PlantUML support `!include` preprocessing https://plantuml.com/preprocessing#393335a6fd28a804

This could help to split complex state design with multiple levels of sub state into multiple smaller PlantUML files.
when generate the code, can use the top level PlantUML file as input

Example:

### `sub.puml`

```
@startuml(id=system) system

[*] -> ACTIVE : POWER_BUTTON
ACTIVE -> IDLE : POWER_BUTTON

@enduml

```

### `top.puml`

```
@startuml test

state POWER_ON {
    !include sub.puml
}

[*] -> POWER_ON
POWER_ON -> [*]

@enduml
```

adamfk, 2023-07-22:
This is a neat idea. Would you be able to help implement?

yongzhy, 2023-07-27:
@adamfk It would be challenge for me. But I could give it a try.
Do you have documentation of StateSmith design itself? 
How to add new features for PlantUML?  Do you have ANTLR4 grammar file for PlantUML? can't find in source repository.

adamfk, 2023-07-27:
Hi @yongzhy. I'll help you implement the feature into StateSmith.

Quick question: is there anyway to get PlantUML to dump the results of its preprocessing to a file? If so, we can just run that result file into StateSmith.

Would something like this work? https://forum.plantuml.net/7944/is-it-possible-to-run-only-the-preprocessor

If the above idea doesn't work or meet your needs, we can get started working on the grammar:

https://github.com/StateSmith/StateSmith/tree/main/antlr

I just pushed a few changes to tidy those scripts up a bit. I'll make a small grammar contributing tutorial this weekend.

yongzhy, 2023-07-31:
hi @adamfk, thanks for point out the `-preproc` option. 
I gave it a try and it solves the issues in this ticket. 
StateSmith not need to add complexity to handle all PlantUML preprocessing features. Just use PlantUML preproc output is the best option. 

Should I close this ticket?

closed 2023-07-31 by adamfk