StateSmith/StateSmith #215: Extend the PlantUML "Ignored diagram elements" list

chriesibaum, 2023-07-27:
Adding the following elements to the "Ignored diagram elements" list would help to draw "nicer" diagrams:

- mainframe 
- skin rose

There might be some more elements to add. Any input is welcome.

adamfk, 2023-07-30:
Hey @chriesibaum! Thanks for volunteering to implement this feature (via discord). It will take a little bit of effort to get all setup, but the actual change should be relatively easy.

I've added more documentation on how to add new features to the PlantUML grammar: https://github.com/StateSmith/StateSmith/wiki/Contributing:-PlantUML

When you fork StateSmith to work on the feature, please start from the `215-plantuml-ignored` branch. I've added some basic [tests](https://github.com/StateSmith/StateSmith/commit/ddde9cafba7bdc9d3ec1819e9894cc40489174dc) (feel free to add more). The tests are failing right now. Your mission 🎯 is to modify the grammar to get those tests passing.

You shouldn't need to write any C# code, but you will still need to build and run the tests.

Ignoring `skin rose` should be super easy I think as it won't clash with other rules.

Ignoring `mainframe` might be a bit more difficult, but I believe you can model it on the existing ignore `scale` rule.

Let me know if you have any problems. Thanks again :)

adamfk, 2023-07-31:
@yongzhy Adding `title` to the grammar should be a small change. If you have time, it would be awesome :tada: if you could implement the `title` ignore feature in the grammar. Why? I'm hoping to grow the StateSmith contributors so that we can keep improving StateSmith 😃 

I'll help support and guide you if you are interested.

yongzhy, 2023-08-01:
@adamfk added new issue for `title`, and will try to work on it.

commit e666c816 (2023-08-09): Merge pull request #224 from chriesibaum/215-plantuml-ignored

added ignore support for plantuml. closes #215
closed 2023-08-10 by adamfk