StateSmith/StateSmith #220: Simply couldn't get it working

lewsut, 2023-08-01:
I'm very interested in this project but the tutorials did nothing for me, on a windows machine I have no idea what's going on, followed your youtube videos with a complete failure. Probably a layer 0 issue, but perhaps consider a set-up video for complete idiots...

adamfk, 2023-08-01:
Can you join the discord group so we can troubleshoot?

adamfk, 2023-08-01:
or can you add some detail of what you tried and what didn't work? I'm not really sure where you got stuck.

Have you tried following the install instructions at https://github.com/StateSmith/tutorial-2/tree/main/lesson-0 ?

If you type in `dotnet --version` on the command line, what do you get?

If you type in `dotnet script --version` on the command line, what do you get?

adamfk, 2023-08-08:
Let me know if you still need help.

closed 2023-08-08 by adamfk

lewsut, 2023-12-18:
Thanks buddy, I was away for a while I shall re try it and report back.