StateSmith/StateSmith #235: C++ logging code not generated properly

Gronk95, 2023-10-31:
The language for the state machine is C++. When enabling logging via $mod / log_r, I get the following errors:

// Step 1: execute action `log_unhandled_event();`
log_unhandled_event(VECU_SM_EventId_any); 

VECU_SM_EventId_any should be VECU_SM_EventId_ANY

Also,
log_unhandled_event() is not generated, I have to insert it by hand.

.csx file is attached.
[VECU_SM.txt](https://github.com/StateSmith/StateSmith/files/13211207/VECU_SM.txt)

Thanks,
Gronk

adamfk, 2023-10-31:
Hi @Gronk95. Do you think you could share the complete diagram with me as well? You can direct message me on discord if you'd like to keep it private.

adamfk, 2023-11-02:
Hi @Gronk95, I like your diagram! It looks great.

Your root state machine has `ANY / log_unhandled_event();`, but no trigger map that defines `ANY`. Without the trigger map, StateSmith thinks you have an event called `ANY`.

This example helps cover trigger maps: https://github.com/StateSmith/StateSmith-examples/tree/main/trigger-map

You need to add something like this:

[image]

> log_unhandled_event() is not generated, I have to insert it by hand.

If I understand correctly, you mean that you have to define the `log_unhandled_event()` function? If so, that is by design. StateSmith can work anywhere with anything. The downside to that is that you have to do a bit of wiring for how you want things logged. Do you want to printf, write to a file, uart...?

Gronk95, 2023-11-02:
Still no joy I'm afraid. I followed the example and included ANY => * in the $Config: TriggerMap. But not I get the following errors from dotnet-script

StateSmith Runner - Compiling file: `VECU_SM.drawio.svg` (no state machine name specified).
Non-related nodes overlap: {id:"226", label:"$CONFIG: Trigge."} with {id:"155", label:"$RenderConfig"}.
DiagramNode
        id: 226
        label: `$CONFIG: TriggerMap
               ANY => *`
        parent.id: 152 (parent follows below)

DiagramNode
        id: 152
        label: `$STATEMACHINE : VECU_SM
               ANY / log_unhandled_event();`
        parent.id: null

You can enable exception detail dumping by setting `RunnerSettings.dumpErrorsToFile` to true.
StateSmith Runner - Finished with failure.

I can't see how to set RunnerSettings.dumpErrorsToFile to true

adamfk, 2023-11-02:
Hi Gronk. This is a new issue. Give this a try: https://github.com/StateSmith/StateSmith/wiki/Troubleshooting#non-related-nodes-overlap

Gronk95, 2023-11-03:
Hi, while I did manage to sort out the overlapping nodes problem. The original bug came back with a vengeance. 

1. The code for the function log_unhandled_event is not generated. That is what I meant when I said I had to generate the code by hand.
2. Each event ID passed to log_unhandled_event ends in lower case when is should be uppercase
3. e.g. log_unhandled_event(**VECU_SM_EventId_abort**) should be log_unhandled_event(**VECU_SM_EventId_ABORT**)

Sorry if I wasn't clear before hand

adamfk, 2023-11-03:
Ah! I'm starting to understand better now. Thanks for explaining. I now see that the `trigger-map` example and https://github.com/StateSmith/StateSmith/issues/45 should be updated to show `CurrentTrigger.ToUpper()`. Thanks!

Note that event IDs are case insensitive in StateSmith. Currently `CurrentTrigger` returns the event ID as lowercase.

If you have setup vscode to provide intellisense to .csx files, you will see a bit of info when you hover over (or navigate to) `CurrentTrigger`:

[image]

Try changing this in your .csx script:

```diff
    public class MyExpansions : UserExpansionScriptBase
    {
        public string log_unhandled_event() => 
-       $"log_unhandled_event(VECU_SM_EventId_{CurrentTrigger})";
+       $"log_unhandled_event(VECU_SM_EventId_{CurrentTrigger.ToUpper()})";
    }
```

I think that will fixes your points 2 & 3 above, but I still don't think I understand point 1.

With the `.ToUpper()` fix above and a TriggerMap, I'm generating what looks like proper C++:
```cpp
static void ROOT_cold_shutdown(VECU_SM* sm)
{
    // ...
    {
        // Step 1: execute action `log_unhandled_event();`
        log_unhandled_event(VECU_SM_EventId_COLD_SHUTDOWN);
        //...
    } // end of behavior for ROOT
}
```

Note that you have to provide the definition of `log_unhandled_event()` somewhere. Maybe put it in `MyLogger.cpp/h` and then include `MyLogger.h` in your generated state machine.

Does that make sense?

Gronk95, 2023-11-03:
Cool.
So the code for log_unhandled_event is not generated by StateSmith?

adamfk, 2023-11-03:
> Cool. So the code for log_unhandled_event is not generated by StateSmith?

That's correct. StateSmith needs the user to provide that because it has zero dependencies.

Here's how it is hand implemented in the trigger map example: https://github.com/StateSmith/StateSmith-examples/blob/main/trigger-map/src/index.js#L20C1-L24

adamfk, 2023-11-03:
How would you like to log the unhandled event? Did you want to printf it? Write it to a file? Send a UART message?

Gronk95, 2023-11-03:
Thanks it all works now.

I'm doing my logging via serial comms messages.

closed 2023-11-03 by adamfk