StateSmith/StateSmith #238: feature - generate SQL

mogul, 2024-01-24:
For example, one might embed a state machine directly into their Postgres/Supabase backend. Reference [this](https://felixge.de/2017/07/27/implementing-state-machines-in-postgresql/) and [this](https://raphael.medaer.me/2019/06/12/pgfsm.html) as HOWTOs.

adamfk, 2024-01-24:
This looks interesting. It might be pretty easy too if it is just generating something like this (from the first link):

```sql
CREATE FUNCTION order_events_transition(state text, event text) RETURNS text
LANGUAGE sql AS
$$
  SELECT CASE state
    WHEN 'start' THEN
      CASE event
        WHEN 'create' THEN 'awaiting_payment'
        ELSE 'error'
      END
    WHEN 'awaiting_payment' THEN
      CASE event
        WHEN 'pay' THEN 'awaiting_shipment'
        WHEN 'cancel' THEN 'canceled'
        ELSE 'error'
      END
    WHEN 'awaiting_shipment' THEN
      CASE event
        WHEN 'cancel' THEN 'awaiting_refund'
        WHEN 'ship' THEN 'shipped'
        ELSE 'error'
      END
    WHEN 'awaiting_refund' THEN
      CASE event
        WHEN 'refund' THEN 'canceled'
        ELSE 'error'
      END
    ELSE 'error'
  END
$$;
```

Is this something you would be interested in helping with?

adamfk, 2024-05-12:
We can re-open if there is community interest in implementing this feature.

closed 2024-05-12 by adamfk