StateSmith/StateSmith #240: Feature request: Parallel state machines with shared event queue

gmkado, 2024-02-15:
Hi @adamfk, I just started playing around with this library and am wondering if its possible to have parallel state machines that share the same event definitions and event queue?

I modified the cpp example to generate two state machines:
[image]

```cs
var settings = new RunnerSettings(diagramFile:  "LightSm.drawio.svg", transpilerId: TranspilerId.C99);
foreach(var name in new[] {"LightSm", "Light2Sm"}) 
{
    settings.stateMachineName = name;
    SmRunner runner = new(settings, new LightSmRenderConfig());
    runner.Run();
}
```

But of course now I have a `Light2Sm_dispatch_event` and Light2Sm_EventId`.  I want them to share the same eventqueue so that eventually events triggered in one state machine will cause transitions in the other.

adamfk, 2024-02-18:
Hello! First of all, apologies for the mix of light and dark themes in the cpp example. I'm fixing now.

draw.io unfortunately doesn't yet have a good solution for switching between light and dark themes. I tend to use dark on all my computers.

It should look like this:
[image]

If you run into this issue again, please let me know so I can correct examples. You can also fix with the right Style & Text tabs:
[image]

StateSmith will eventually support concurrent/parallel/orthogonal state machines: https://github.com/StateSmith/StateSmith/issues/75

In the mean time, I do have an example of how you can use one state machine to trigger events in another state machine. It's an older example and uses the yEd file format (before we switched to draw.io). You can see it in operation here: https://www.youtube.com/watch?v=9czSDothuzM and play with the code in an online simulator here: https://wokwi.com/projects/351165738904453719

The general idea is that each button state machine has a number of bitfields for "output events":
```
        uint16_t output_event_press : 1; // output
        uint16_t output_event_release : 1; // output
        uint16_t output_event_held : 1; // output
        uint16_t output_event_tap : 1; // output
```
When a button state machine determines that the button was held, it sets its `output_event_held` bitfield to true.

The App.c file connects the button state machines to a larger laser tag menu state machine. See function `dispatch_button_events(void)`. It essentially looks at each button's "output" variables and converts them to events for the laser tag state machine.

Does that make sense?

gmkado, 2024-02-18:
Yep that makes sense, thanks for the example!  I'll close this in favor of #75

closed 2024-02-18 by gmkado