StateSmith/StateSmith #241: Non-related nodes overlap in Tutorial-2 Lesson 1

emmby, 2024-04-22:
I'm trying to follow Tutorial-2 Lesson 1 to create an ON2 node.

However, when I run `dotnet-script code_gen.csx` I get the following error
```
mike@mikes-mbp lesson-1 % dotnet-script code_gen.csx

StateSmith Runner - Compiling file: `LightSm.drawio.svg` (no state machine name specified).
![LightSm drawio](https://github.com/StateSmith/StateSmith/assets/310880/cb9f634a-9e7d-460f-92bc-c1a3eddff4f7): {id:"79", label:"ON2\\nenter / win…"} with {id:"12", label:"$STATEMACHINE :…"}.
DiagramNode
        id: 79
        label: `ON2
               enter / window.alert("Light is ON");`
        parent.id: null

You can enable exception detail dumping by setting `RunnerSettings.dumpErrorsToFile` to true.
StateSmith Runner - Finished with failure.
```

SVG: 
[image]

adamfk, 2024-04-22:
Hi @emmby. Thanks for reporting the issue! I'll take a look at it in about an hour or so (during a break). Have you already joined the discord group? Someone might be able to help you there sooner.

emmby, 2024-04-22:
It appears that the problem is that you must enter the group before you can make the change. If you don't enter the group, you get the error.

I'll submit a PR to update the readme.

adamfk, 2024-04-22:
ah. This is one of the gotchas with using draw.io. It allows drawing images that look like they are grouped, but are not. See https://github.com/StateSmith/StateSmith/wiki/Troubleshooting#non-related-nodes-overlap

I just made the troubleshooting section more prominent on the main wiki page, but I also like your suggestion of adding text to the tutorial. Great idea. Looking forward to the PR :)

emmby, 2024-04-22:
https://github.com/StateSmith/tutorial-2/pull/2

adamfk, 2024-04-24:
Related to https://github.com/StateSmith/StateSmith/issues/81

StateSmith version 0.9.10 adds a help URL to the exception:
```
Non-related nodes overlap: {id:"37", label:"NOT_IN_GROUP\\nen…"} with {id:"27", label:"$STATEMACHINE :…"}.
See https://github.com/StateSmith/StateSmith/issues/81 for how to fix.
```

Feel free to re-open the issue if there's something else you think would be helpful.

closed 2024-04-24 by adamfk