StateSmith/StateSmith #260: binaries build

totocaca123, 2024-05-05:
Currently, we have to download source code and build dlls.
It could be possible to add a workflow to add zip binaries for different targets
ubuntu, macos, windows, et net6/7/8 (maybe remove 7 because it's not a LTS)

adamfk, 2024-05-07:
I would like to add binaries, but I'm not sure of the best way to do that.

Would we bundle the dotnet runtime with StateSmith?

I'm really curious to see if we could create native AOT binaries: https://learn.microsoft.com/en-us/dotnet/core/deploying/native-aot/?tabs=net7%2Cwindows It would be nice to provide binaries that aren't 100+ MB.

# Expansions would need to change
One current challenge is that the StateSmith expansions come from C# code in .csx files. We rely on an external dotnet sdk to compile those expansions. We could use Roslyn to compile them. 

There's an idea for this here: https://github.com/StateSmith/StateSmith/issues/118

And another related idea here: https://github.com/StateSmith/StateSmith/issues/264

totocaca123, 2024-05-07:
To run cli we need a csx file. Ideally, I want to download binaries, with dll and having an exe file to run with bash, cmd command

For AOT, .net6 seems to be not compatible

adamfk, 2024-05-10:
I'll try to add environmental variables in about a week or so. Will report back here when ready for more testing.

adamfk, 2024-08-19:
closing for now. We can re-open or make a new one as feedback/issues arise.

closed 2024-08-19 by adamfk