StateSmith/StateSmith #261: test on windows fails

totocaca123, 2024-05-05:
If fails locally or with github actions.
Build is OK but some tests requires gcc compilation and dotnet.

WSL is used on windows, but I expect to have powershell (as dll generated cannot run under wsl)

ubuntu and macos tests are ok as it uses bash

adamfk, 2024-05-06:
I think when we first added github actions to StateSmith, the only free OS was linux. Is it possible to add `windows-latest` for free now?

The specification test running is rather complicated. It's written in a way to try and make it as easy as possible to add new programming languages to StateSmith. It could definitely use some clean up and refactoring.

> WSL is used on windows, but I expect to have powershell (as dll generated cannot run under wsl)

WSL is only used to build and run the generated code projects:

1. [native OS] the StateSmithTest project is built and run
2. [native OS] each supported programming language runs StateSmith to generate target files (.c, .h, .cs, .js...). This is only done once per programming language. We use an xUnit IClassFixture for this. For C#, see https://github.com/StateSmith/StateSmith/tree/main/src/StateSmithTest/spec/2/CSharp .
3. An external process is then invoked to build the C/C#/JS specification project. If on Windows, WSL2 is used. The C# specification test project is here: https://github.com/StateSmith/StateSmith/tree/main/src/test-misc/csharp-spec2
4. For each of the [46 individual specification tests](https://github.com/StateSmith/StateSmith/blob/main/src/StateSmithTest/spec/2/Spec2Tests.cs), an external process is invoked to run the C/C#/JS specification projects. If on Windows, WSL2 is used. The specification project run is provided input arguments for how to exercise the state machine. As the state machine is exercised, it prints to std out. The process output is captured and compared to what we expect.

Does that help explain why and when we use WSL?

This is pretty complicated. I think I'll make a write up or video that explains it better.

adamfk, 2024-05-07:
If you run into issues using StateSmith, you can sometimes find the answer on the wiki: https://github.com/StateSmith/StateSmith/wiki/Building-Source-and-Running-Tests

I like the idea of also running unit tests with native windows programs. Let's continue that particular discussion here: https://github.com/StateSmith/StateSmith/issues/262

adamfk, 2024-05-31:
@totocaca123 Do you think there is more work to do here? I'm pretty happy with the compiler test changes implemented in #262.