StateSmith/StateSmith #262: Question: compiler use

totocaca123, 2024-05-05:
Currently the project use gcc. For github action, it requires to install on server a compiler. The best would be to use a C/Cpp compiler compatible for windows and ubuntu (gcc, llvm or microsoft compiler) and compatible with current code

adamfk, 2024-05-07:
I like the idea of using native Windows programs to run unit tests when on Windows. Not only would it make life easier for Windows users contributing, it would also let us test on another compiler other than gcc.

Which C++ compiler would you suggest we use? MSVC?

It looks like this is already possible: https://stackoverflow.com/a/77140873/7331858

Thoughts?

totocaca123, 2024-05-07:
For the following compilers, I know it well:

- gcc : available on windows, ubuntu, macos
- llvm (is my favorite compiler) is available on win/ubuntu/llvm
- MSVC: i don't know it very well , only advantage is it inclusion with microsoft env

Moreover, a choice has to be done on cpu target (x86 or x64) as it impact some datatype size in C code

adamfk, 2024-05-08:
What if we make some environment variables so that it is easy to use any compiler we want?

* `SS_TEST_C_COMP` - path to compiler to use for C99 (`gcc`, `clang`, `cl`, ...)
* `SS_TEST_C_COMP_ARGS` - arguments like `-Wall -std=c11`
* `SS_TEST_CPP_COMP` - path to compiler to use for C++ (`g++`, `clang++`, ...)
* ...

The StateSmith specification test runner will use those if found.

This would make it relatively easy to run the tests for each compiler.

Thoughts?

adamfk, 2024-05-11:
I will hopefully have something for us to try in a week or so. Thanks for the suggestions.

adamfk, 2024-05-14:
Good news! Tests now run significantly faster on my Winows machine :) WSL is only used for C compiler when needed. We don't need to install nodejs and dotnet into WSL for testing (StateSmith will look for them to be installed normally). Also removed dependency on net6.0 for tests. We also have more flexibility with environment variables.

Top of page is now updated with this info: https://github.com/StateSmith/StateSmith/wiki/Building-Source-and-Running-Tests

In latest main branch, we can specify environment variable `SST_C_COMP_ID` to one of:
* GCC (unix only)
* WSL_GCC (windows only)
* CLANG (unix only for now)
* MSVC (wired up but [doesn't work yet](https://github.com/StateSmith/StateSmith/blob/66815f9721d17904111810d30539acdc4aa60e56/src/StateSmithTest/Processes/CComp/MsvcCompilation.cs#L32-L34))

Usage on Unix:
```
$ export SST_C_COMP_ID="clang"
$ dotnet test
```

Windows:
* either set `User variable` in GUI (not machine section)
* or can set with process variable `> set SST_C_COMP_ID=GCC` `dotnet test`

@totocaca123 are you able to give it a try?