StateSmith/StateSmith #264: How might we make StateSmith “zero config” for most models?

emmby, 2024-05-05:
How might we make StateSmith onboarding easier for new users?

StateSmith is a powerful tool but the learning curve is steep relative to its commercial peers. Some of this curve can be addressed by better tooling such as https://github.com/StateSmith/StateSmith/issues/249 and some of it can be addressed with additional documentation, but much of the complexity comes from the C# configuration script and the manual Expansions mapping needed to configure every model and manually map model guards and actions to code in the destination language.

The mapping support is very powerful and customizable, but it introduces additional complexity to every model. Learning enough C# to get by can be a headwind for a bootstrapping new user. Understanding how the configuration and Expansions work is unique to StateSmith and can be confusing. Expansions can be difficult to make cross-platform. And because the system is so flexible, there are often multiple ways to solve a problem and choosing between them can be subtle and require experience.

This complexity contributes to StateSmiths’ ramp-up time, and it also contributes to additional complexity in long term maintenance.

**Proposal: Provide a standardized way to map guards and actions to any supported language, so that most state machines require zero-configuration to generate code.**

## Grammar

StateSmith already provides “zero-config” for state triggers, eg. enter / exit / do. Triggering on enter/exit/do does not require any additional configuration. 

Guards and actions can also be made zero-config with a simple grammar. For example,

```jsx
enter [isReady] / doTheThing()
```

Guards like `isReady` must evaluate to a boolean. They can reference variables (eg. `isReady`) or functions (eg. `isReady()`), and can leverage simple boolean operators such as `&&`, `||`, etc. 

Similarly, Actions can be a sequence of statements that reference variables, functions and return values, simple operators, and statements separated by semicolons. Examples might include:

```jsx
enter [isReady] / {
  doTheFirstThing();
  doTheSecondThing(counter++) || abort()
}
```

TODO simple Backus-Naur grammar specification

## Developer experience

Generating code from these models would not require any configuration. It would not require any knowledge of C# or StateSmith’s configuration language.

The user would trigger the codegen, and out would pop implementations appropriate to their target language. The user would be responsible to implement the necessary interfaces.

For example, a C++ generated interface for the above might look something like:

```cpp
/* generated gen/MySMInterface.hpp */ 

class MySMInterface {
public:
  virtual bool getIsReady() = 0;
  virtual void setIsReady(bool b) = 0;
  virtual int getCounter() = 0;
  virtual void setCounter(int) = 0;
  
  virtual void doTheThing() = 0;
  virtual bool doTheSecondThing(int counter) = 0;
}
```

and the user’s implementation would be:

```cpp
#include "gen/MySmInterface.hpp"

class MySMImplementation : public MySmInterface {
public:
  virtual bool getIsReady() { ... }
  virtual void setIsReady(bool b) { ... }
  virtual int getCounter() { ... }
  virtual void setCounter(int) { ... }
  
  virtual void doTheThing() { ... }
  virtual bool doTheSecondThing(int counter) { ... }
[... 96 lines cut ...]
It’s best practice to implement automated testing for your state machines. These tests are typically dependent only on the generated state machine itself and not on any developer-supplied code (which might be designed to run in environments such as embedded devices which may be hard to emultated on a buildbox).

With zero-config, StateSmith can automatically generate all the associated test scaffolding and sample tests you might need for your project. These tests could easily be implemented in the same language as your target (eg. C99 if you’re building for an embedded device), or they could just as easily be implemented in another language such as javascript, which might be more attractive due to its rich test library support and quick build/test cycle. The developer can choose, StateSmith will generate both by default.

### Automated syncing of generated files on save

In StateSmith today, saving the image file and generating the code are two separate operations. The user must first save, and then run a csx file to generate the code. The tool does not necessarily know which config to use, so it cannot generate the files.

With zero-config, the second step will no longer be necessary. Because most models will not need a config file, the drawio plugin can simple run the default codegen automatically after every save, thereby keeping the generated files in sync with the svg on every save.

### Generate other niceties by default

Zero-config opens up the possibility of auto-generating other niceties as well, such as automated FSM tracing. This would be enabled be default, and easily bypassed/disabled.

### Easier sharing of models in the StateSmith community.

Because every model generates a working JS implementation, it will be easier for users to share models in the StateSmith community. Models could be shared and posted to the StateSmith examples directory without modification (and without needing to include any environment-specific implementation code), and interacted with on the StateSmith website.

## Critical User Journeys

- As a new user, the tutorial should show me how to create a new FSM from scratch (not cloned) and interact with it in a web browser without running any commands (other than setup) manually.
- As a new user, the tutorial should show me how to make changes to that FSM, save them, and see them reflected in the web browser on reload.
- As a new user, the tutorial should show me how to add C++, C#, or C codegen output with a one-ish line config or settings change. That output should include sensible defaults like optional state transition logging, assertions, interfaces to reflect user supplied code, etc.
- As a new user, I want any language interactions I reference from my FSM to be cross-platform by default and require zero config. The tool/compiler should clearly indicate which methods/functions I need to implement/subclass to complete the implementation of my FSM
- As a new user, the generated code should include clear instructions in the comments about how to use it, eg. how to create the main file, implement the functions, etc.

adamfk, 2024-05-08:
Lot's of good ideas in here. Thank you very much for writing them up :)

I'm starting to break individual ideas out to separate issues for discussion.

---

It would be great to not need a separate .csx file. Doing everything the diagram would be very nice. Although, it might only apply to draw.io diagrams. Not sure how we would add that to plantuml (special comment blocks?).

The main reason we need the .csx file is for expansions. Most (all?) of the Render Config settings can already be specified in the diagram: https://github.com/StateSmith/StateSmith/wiki/Diagram-Based-Render-Config This probably needs to be more visible.

That said, not all languages need expansions. C# doesn't need them at all. You simply specify the base class the generated state machine will extend. See https://github.com/StateSmith/tutorial-2/blob/main/lesson-5/c%23/LightSm.csx#L42

JavaScript can do something similar, but you still need to type `this.` to access inherited members. This is kind of annoying. We could implement an a default expansion that would apply `this.` to everything so that `count++` in the diagram becomes `this.count++`. Would need some black listing though so that `console.log()` doesn't end up as `this.console.log()`.

Even C and C++ don't actually need expansions if the generated .c/.cpp file is #included into the user main file. Would change the .c file extension to be .inc then. https://github.com/StateSmith/StateSmith-examples/issues/2

---

Interim idea:

Using your interface concept and a default expansion could work for any language. Exciting! A state machine would have a field to a user defined class/struct that has everything the state machine needs to do.

```c
typedef struct LightSm_Vars
{
    LightSmInterface iface;
} LightSm_Vars;
```

The user then writes the `LightSmInterface` class in their regular editor/language (not in csx file and not in the diagram). If a user forgets to add something to their interface class, their compiler will warn them. Unless they are using something like loose like Python without an analyzer.

Inside the diagram they define a default expansion that maps something like `count++` to `sm->vars.iface.count++`. Or `do_something(count)` to `sm->vars.iface.do_something(sm->vars.iface.count)`. If a design wants to call a global function, the user can either add it to their interface class, or add an expansion (in the diagram) that calls the global function.

This actually feels very possible and maybe not too much work. It would be really nice to have a single diagram solution (no .csx required). Then we could distribute StateSmith.Cli as a native binary and users wouldn't need to install dotnet at all unless they want to do something fancy.

---

I'm not quite sure about automatically inferring interfaces. Personally, I think I would prefer to manually specify the interface so that I know for sure that it is correct and doesn't have a subtle mistake (wrong int type or something). However, it would be really handy if StateSmith would infer the interface and provide it as a suggestion to the user as a starting point. That could be really nice.

---

> Zero-config opens up the possibility of auto-generating other niceties as well, such as automated FSM tracing

We can do that now with .csx scripts a few different ways. Here's one: https://github.com/StateSmith/StateSmith-examples/tree/main/logging-simple-script-2

It would be nice to retain powerful features like this when we support a single diagram file though. We could probably just standardize on tracing functionality and then add settings nodes to the diagram like RenderConfig nodes.

emmby, 2024-05-08:
Love it! I think the interim idea would go a long way to making things easier to use, and agree it sounds simpler to implement. The idea of specifying the interface in the target language rather than in the diagram makes a lot of sense, and it also addresses the plantuml issue.