StateSmith/StateSmith #265: feature - automated code generation when diagram saved

adamfk, 2024-05-08:
From https://github.com/StateSmith/StateSmith/issues/264:

> In StateSmith today, saving the image file and generating the code are two separate operations. The user must first save, and then run a csx file to generate the code. The tool does not necessarily know which config to use, so it cannot generate the files.

> With zero-config, the second step will no longer be necessary. Because most models will not need a config file, the drawio plugin can simple run the default codegen automatically after every save, thereby keeping the generated files in sync with the svg on every save.

I really like this idea, but it will take some time to get to "zero-config".

A very nice interim step could be to add "watch" functionality to the StateSmith.Cli tool. Something kind of like the watch command for typescript. Watch these diagram/csx files and run code generation when they change.

Could be something like `ss.cli watch` and use the statesmith.cli.json manifest file or work recursively in the current directory.

Might not even be that hard as ss.cli already does a good chunk of that work.

emmby, 2024-05-08:
That makes a lot of sense

adamfk, 2024-06-03:
Let's take a look at this after #285 is finished.

commit fa16b994 (2024-10-09): cli - `watch` working well. Lowered CPU a bit and tidy. Close #265
closed 2024-10-09 by adamfk