StateSmith/StateSmith #266: feature - generate test scaffolding by default

adamfk, 2024-05-08:
From https://github.com/StateSmith/StateSmith/issues/264:

> It’s best practice to implement automated testing for your state machines. These tests are typically dependent only on the generated state machine itself and not on any developer-supplied code (which might be designed to run in environments such as embedded devices which may be hard to emultated on a buildbox).

> With zero-config, StateSmith can automatically generate all the associated test scaffolding and sample tests you might need for your project. These tests could easily be implemented in the same language as your target (eg. C99 if you’re building for an embedded device), or they could just as easily be implemented in another language such as javascript, which might be more attractive due to its rich test library support and quick build/test cycle. The developer can choose, StateSmith will generate both by default.

This is a powerful idea, but I'm not sure how we would go about doing it.

I'd thought of starting with something simpler though:
* auto generate all the test code that fully exercises a state machine, but it relies on user code to satisfy transitions with guard clauses.
* it could generate all the stub transition functions for you though.

Consider testing this SM: https://github.com/StateSmith/tutorial-2/blob/main/lesson-5/js-web/LightSm.drawio.svg

It would generate a test case for every single transition in the state machine.

The only transition it can't figure out how to take is from ON2 to ON_HOT. It won't be smart enough to satisfy the guard `[count >= 3]`. But it could create a stub function for the user:

```c
void ss_test_satisfy_guard_from_ON2_to_ON_HOT(LightSm * sm, MaybeSomeContext ctx)
{
  // user code either directly makes `count >= 3` (white box testing) 
  // or has the state machine transition back and forth to get `count` large enough naturally (black box testing).
}
```

The generated test code would get to ON2 (from start) and then call that user function. It would then verify that it ended up in `ON_HOT`.

There's a fair amount of work here. Also, how would we pick a test framework to use? Each language has many. Would we try to abstract away test framework details? Roll our own mini test framework per language? Some interesting challenges :)

It would be really sweet if we could leverage the transpiler so that each language could generate tests without a ton of implementation effort.

emmby, 2024-05-08:
Here's how another tool does it: https://www.itemis.com/en/products/itemis-create/documentation/user-guide/sctunit_sctunit_by_example

They have their own junit-like test language that lets you do things to your state machine and test verify that states and transitions. eg. 
```
testclass light_switch_tests for statechart light_switch {

    @Test
    operation isStateOffActiveAfterInit () {
        enter
        assert active (light_switch.main_region.Off)
    }

    @Test
    operation isStateOnActiveAfterOperateInOffState () {
        enter
        raise operate
        proceed 1 cycle
        assert active (light_switch.main_region.On)
    }
}
```

I don't think there's a lot of value in inventing a test language, so I was thinking that SS would implement certain primitive operations like enter(), raise(), proceed(), assert(), and then provide implementations of those methods in js/c/c#/etc. That way, users would be able to test in their preferred language.

I think roughly :hand-wave: the work would be something like:
1. generate hooks into the FSM to allow testing
2. Generate a testing interface in each language that provides the above operations using those hooks

That would be enough for people to test with their language and framework of choice. Once you have that, then you could provide some additional niceties like generating default scaffolding, etc., but agree that comes with many design choices.

adamfk, 2024-05-09:
I created a small sample project that shows how we can hook into StateSmith to generate test plans, test scaffolding, or documentation: https://github.com/StateSmith/StateSmith-examples/tree/main/test-gen-0

This is one of the parts that I do love about csx files. It is super easy to prototype ideas and easily get high fidelity model information from StateSmith (or even mod it before code gen). No need to modify StateSmith at all (in most cases).

emmby, 2024-05-11:
Here's an example from https://github.com/StateSmith/StateSmith-examples/pull/3

```js
// Example test file using jest. User-supplied, not generated

import {LightSm} from './LightSm.js';
import {jest} from '@jest/globals'; // We'll use jest in this example. see package.json for config details

// Mock the functions used by the state machine
// We recommend mocking rather than importing your actual functions,
// to keep these tests purely about testing the state machine itself.
// Your implementations should also be tested, but in separate tests.
globalThis.println = jest.fn();
globalThis.light_blue = jest.fn();
globalThis.light_yellow = jest.fn();
globalThis.light_red = jest.fn();

beforeEach(() => {
    jest.clearAllMocks();
});

test('starts in the off state', () => {
    const sm = new LightSm();
    sm.start();
    expect(sm.stateId).toBe(LightSm.StateId.OFF);
});

test('println to be called once on startup', () => {
    const sm = new LightSm();
    sm.start();
    expect(globalThis.println.mock.calls).toHaveLength(1);
});

test('light is blue when turned on', () => {
    // Arrange
    const sm = new LightSm();
    sm.start();

    // Act
    sm.dispatchEvent(LightSm.EventId.INCREASE);

    // Assert
    expect(sm.stateId).toBe(LightSm.StateId.ON1);
    expect(globalThis.light_blue.mock.calls).toHaveLength(1);
});

test('light can be turned off', () => {
    // Arrange
    const sm = new LightSm();
    sm.start();
    sm.dispatchEvent(LightSm.EventId.INCREASE);
    expect(sm.stateId).toBe(LightSm.StateId.ON1);

    // Act
    sm.dispatchEvent(LightSm.EventId.DIM);

    // Assert
    expect(sm.stateId).toBe(LightSm.StateId.OFF);
});

test('count how many times INCREASE is pressed in ON2', ()=>{
    // Arrange
    const sm = new LightSm();
    sm.start();
    sm.dispatchEvent(LightSm.EventId.INCREASE);
    sm.dispatchEvent(LightSm.EventId.INCREASE);
    expect(sm.stateId).toBe(LightSm.StateId.ON2);
    expect(sm.vars.count).toBe(0);

    // Act
    sm.dispatchEvent(LightSm.EventId.INCREASE);

    // Assert
    expect(sm.stateId).toBe(LightSm.StateId.ON2);
    expect(sm.vars.count).toBe(1);
})

test('increase until red', ()=>{
    // Arrange
    const sm = new LightSm();
    sm.start();
    sm.dispatchEvent(LightSm.EventId.INCREASE);
    sm.dispatchEvent(LightSm.EventId.INCREASE);
    sm.dispatchEvent(LightSm.EventId.INCREASE);
    sm.dispatchEvent(LightSm.EventId.INCREASE);
    expect(sm.stateId).toBe(LightSm.StateId.ON2);
    expect(sm.vars.count).toBe(2);

    // Act
    sm.dispatchEvent(LightSm.EventId.INCREASE);

    // Assert
    expect(sm.stateId).toBe(LightSm.StateId.ON_HOT);
    expect(sm.vars.count).toBe(3);
})

/// Notes:
/// - in JS we have access to the internals of the state machine so we don't need
///   to implement a special test api. For other languages we'll need a test api.
```

emmby, 2024-05-14:
At a high-level, would the approach be something like:

- clone [JavaScriptGilVisitor.cs](https://github.com/StateSmith/StateSmith/blob/main/src/StateSmith/Output/Gil/JavaScript/JavaScriptGilVisitor.cs) to JavaScriptTestGilVisitor.cs
- JavaScriptTestGilVisitor.cs would be mostly static boilerplate, except it would need to identify method declarations in actions and generate fakes for them. Would that be VisitMethodDeclaration in the visitor, or something else like Behavior.cs/actionCode?
- add JavaScriptTestGilVisitor.cs to GilToJavaScript.cs  so the test samples get created automatically during js codegen

adamfk, 2024-05-14:
@emmby  I just made a new repo and made you a collaborator. I copied in your example code and made an "adam" branch where I can prototype ideas. You can make a "mike" branch if you like or use main.

I'd like to do the code generation here for now: https://github.com/StateSmith/exp-test-gen/blob/main/src/LightSm.csx#L28-L60

1. Instead of it printing to console, it could generate the test code files. It isn't the long term solution, but we can quickly prototype there quite easily.
2. We get something very basic working. The simplest useful thing. 
3. Celebrate! 🎉
4. We do something a bit bigger.
5. We consider our approach. Can we make it language/framework agnostic enough that the generated test code could be transpiled with StateSmith?
6. ...

adamfk, 2024-05-14:
this is the kind of thing I was hoping StateSmith could generate. The concept isn't fully finished, but this should hopefully give an idea: https://github.com/StateSmith/exp-test-gen/blob/adam/src/LightSm.jest.test.js#L16-L55

Note that the helpers build off previous ones.

emmby, 2024-05-15:
I added an example with some questions here: https://github.com/StateSmith/exp-test-gen/pull/1

The helpers are some interesting syntactic sugar. Seems like they could get pretty complex pretty quickly if there are multiple paths to getting to different states.  Also, some of them might be impossible to generate, for example if any of the transitions depend on user-implemented guards/actions.

I feel pretty comfortable arranging my initial states manually in the test cases, but maybe there are cases where auto generating these helpers would be useful. That being said, seems like more of a nice to have than essential for testing.

adamfk, 2024-05-15:
Often at my work, we want to test every single transition in a state machine to make sure it behaves as expected. When there are 30+ states, it can become a lot of work. Not hard work, but still a lot of work.

One of the reasons for the helper functions (even if handwritten) is so that the test code remains easy to maintain. If we are testing a 30 state FSM and we duplicate the Arrange code for each test, it becomes a lot of work when something changes in the state machine in one of the first states.

With the helper methods, we only need to update `start_to_on1()` if the diagram changes how it transitions from OFF to ON1. Just one method to update. 

However, if we repeat the Arrange code in X tests, then we have to update each and every one of them which quickly becomes a chore. Let's assume we have a 10 state FSM where it goes from state 1 to state 10. All based on events. 

To go from start to state 1 we do:
```
sm.start();
```

To go from state 1 to state 2 we do:
```
sm.start();
sm.dispatchEvent(LightSm.EventId.INCREASE);
```

To go from state 2 to state 3 we do:
```
sm.start();
sm.dispatchEvent(LightSm.EventId.INCREASE);
sm.dispatchEvent(LightSm.EventId.INCREASE);
```

In terms of repeated lines of code, we have `sum = n * (n+1)/2` where `n` is the number of states. If we have 10 states, we have `10 * (10 + 1) / 2` = 55 lines of repeated code. This grows exponentially though with the number of states. If we have 30 states, we have `30*31/2 = 465` lines of repeated code. Does that make sense? It's late and my math could be off 🤪 It becomes too much to hand code and repeat everything.

You are correct though, there are certain transitions that StateSmith won't be smart enough to figure out. It can figure out simple event based transitions, but not guards like `[ count >= 3]`.

[image]

That said, I'd still welcome generated code that relies on user code to satisfy those guards. If nothing else, it would provide a good skeleton and the user will hopefully just need to add some ribs to help it along. Maybe I should make a separate issue for that though. I got a bit too excited and blew up the scope of this issue :) It would be very handy for my work.

So for this issue, the remaining work is to be able to detect all the functions called and variables accessed in a diagram so that they can be automatically mocked. Right?

adamfk, 2024-05-15:
While that larger/harder task is tricky, I think it would be an incredible help for medium sized state machines. A massive bonus and time saver.

adamfk, 2024-05-15:
Just added the ability to detect variable and function use inside state machine diagrams: https://github.com/StateSmith/StateSmith/issues/275

Will update example with how to use.

emmby, 2024-05-15:
Got it, totally makes sense how that would be valuable for larger models.
IExpander looks perfect, will take a look. ty!

adamfk, 2024-05-15:
Added a PR into your PR: https://github.com/StateSmith/exp-test-gen/pull/2/files

It prints this:
Diagram accesses these variables:
  count
Diagram accesses these functions:
  println
  light_red
  light_blue
  light_yellow

adamfk, 2024-05-17:
@emmby I took your work from [our experiment repo](https://github.com/StateSmith/exp-test-gen/blob/a0b7e0d332e6a7f34d1a5dc51e09f9e580e702bc/src/LightSm.csx) and slammed it into the CLI.

There's a lot of cleanup to do, but it allows us to quickly test whether we like this or not.

Here's your generator: https://github.com/StateSmith/StateSmith/compare/main...266-cli-test-scaffold#diff-7309c7fdfa59c078ea8110df2f29b740d2dc05b96a52b5f84bf2f7c16618c239

Here's simple tests for your generator: https://github.com/StateSmith/StateSmith/compare/main...266-cli-test-scaffold#diff-7309c7fdfa59c078ea8110df2f29b740d2dc05b96a52b5f84bf2f7c16618c239

To test it out:
* clone the main StateSmith repo and checkout the `266-cli-test-scaffold` branch.
* cd to `StateSmith/src/StateSmith.Cli` directory
* run command `dotnet run --framework net8.0` or use whatever framework you have installed.
* It will run the CLI.
* `dotnet run` is slow for iterative testing when you haven't made any code changes as it does a bunch of work to see if it needs to recompile. You can run faster without building with this command: `dotnet run --framework net8.0 --no-build --no-restore`

Make sure you select JavaScript when creating your new project or it won't show option to generate test scaffold.

[image]

Let me know what you think. It's not smart enough to do automated mocking yet.

emmby, 2024-05-18:
Very cool! It generated a sample svg and the sample tests. I couldn't figure out how to generate a test file for my BeadSorter.drawio.svg, though.

Would it be more valuable to tie the test generation to the code generation rather than the project setup? Project setup is only done once, which means you can only generate test files for the example SVG and never for your own SVG. As your SVG evolves, there's no way to update the test example and get updated mocks or helpers, for example. But if we tie it to the code generation step, then it will always be generated from your actual SVG.

adamfk, 2024-05-21:
I agree. Creating test scaffolding only at startup is a very simple starting step. It doesn't help our users with their existing designs (assuming they haven't already written tests for them).

There are a number of milestones for this test generation goal:

### Level 1 - Ideal - everything auto generated
* all dependencies auto mocked
* all test code auto generated
* unlocks simulation capability

### Level 2 - Great - most auto generated
* most dependencies auto mocked
* most test code for transitions auto generated. requires some user help.
* could unlock simulation capability with some user work

### Level 3 - Very Good - some auto gen
* some dependencies auto mocked
* some initial test code auto generated

### Level 4 - Good - auto gen at project creation
* no dependencies mocked
* basic template setup

We can do level 4 easy. Level 1 is a fair amount of work and may require restrictions on what user designs can do. Inline action code like `SomeClass::DoStuff(...)` might not be allowed initially.

My understanding is that we want to target Level 4&3 now. Other levels in future [like I mentioned above](https://github.com/StateSmith/StateSmith/issues/266#issuecomment-2111494919).

I'm not entirely sure about the workflow for level 2&3 where part of the test code is generated and part of it is hand written. A few ideas of the top of my head:
* File Option 1 - we auto generate separate test helper files for each fsm. The user would write their own `LightSm.test.cpp/hpp` file and we would generate `LightSm.test-helper.cpp/hpp` file(s).
* File Option 2 - we dynamically inject/update special sections within user handwritten test file. Have to do this carefully to avoid user rage :) Single test file `LightSm.test.cpp`.
* File Option 3 - we print/create example test files that users can copy from.

### File Option 1 - auto generate separate test helper files for each fsm
Pros:
* will work well initially for Level 3 (some)
* no collision between user and autogenerated files

Cons:
* will become harder to work with as we approach Level 2 (user and auto generated code being co-dependent). It has to be done carefully to avoid include circular dependencies. Or it requires additional files which become cumbersome to work with (4+ files to test a single fsm in c/c++)
* May require additional special instructions/arguments to test generator so that generated code can compile.

### File Option 2 - shared single test file
Pros:
* gives us the largest long term potential as it is the most powerful and most flexible approach.
* It easily allows for user and auto generated code being co-dependent
* I also like the single test file per fsm. My one coworker really doesn't like lots of files :)
* can work roughly the same for any language (doesn't require preprocessor or modules...)

Cons:
* The main downside is that we need to coordinate writing to the single "shared" test file with the user. If we write to that file every time we run code gen, the chances are pretty high that we will cause some user frustration. I'd like the user to request the test code to be updated to help avoid this frustration.
* Might be annoying to review in pull requests. Large amount of auto generated code mixed in with user code.

### File Option 3 - print/create example test files
I think I'm the least clear about this workflow, and I believe this is our current plan. My understanding below.

* I have an existing fsm design (`LightSm`) and am ready to do some testing
* I run code gen and it generates `LightSm.example.test.js`. It includes some mocks for me.
* I copy, move and rename the example test file and make it work for my design
* I continue to manually modify/tweak my fsm. I run the code generation and make multiple commits.
* In my above commits, I used another function `foo()` in my fsm. `foo()` isn't mocked in my test.js file yet, but it is in the example.test.js file.
* I run the tests and they fail because `foo()` isn't mocked.
* I troubleshoot.
* I either manually add `foo()` to my mocks in my test file, or I find the example.test.js and copy more from it into my real test.js file. Copying from the example file would be helpful if many new functions need to be mocked.

Is that correct? Maybe you could write a few small scenarios of how users would use this? I feel like I'm might be missing something here.

Thanks!

emmby, 2024-05-21:
Right, so there's a lot in there, let me try to unpack a few things.

Yes, I agree I'm mostly focusing on File Option 3. I think the value that StateSmith can provide for tests is:

1. Provide just enough to get people started. This means providing a single test that works out of the box and some examples of how to write more meaningful tests. It means providing them with some default mocks so they can see how the patten works and modify it as needed. It may also mean providing some nice helpers to navigating the state diagram.
2. We should do the above regardless of whether they're starting from scratch or iterating on an existing model. Maybe they already have a bunch of tests, or maybe they don't have any, but we should still provide good examples to build on.
3. We should do it "on by default." Many users may never end up writing tests, but nobody should ever be able to say that StateSmith made it too hard for them to get started :)
4. The examples should always be up to date and working. If users make changes to their SMs, the examples will reflect that automatically. We may not be able to keep the tests themselves up to date without introducing a lot of complexity, but the examples is really all the users need to get started.
5. Introduce them to the idea that FSMs should be tested, and that if they want to it's kind of easiest to do it in JS, regardless of what their target language is. 

I think the following are non-goals:

- I don't think StateSmith should try to do anything too fancy like auto-generate a bunch of tests. I don't think there's a lot of value here.
- I don't think StateSmith should be expected to keep the tests themselves up to date in a meaningful way. I think the best we can do is keep the examples up to date for the user to be able to crib off of. They can always refer back to the examples and know that they are 100% up to date and working.

My thinking on File Option 2 (dynamically inject/update special sections within user handwritten test file) is that it sounds complex both for the user and the StateSmith maintainers. I also think it's targeting more advanced users/FSMs, which I don't think is the goal. I think it's better to do something simple here that anyone can understand and copy at a glance than to try to be sophisticated to avoid copying. At least for a v1.

File Option 1 (auto generate separate test helper files for each fsm) is something we may want to do regardless of which direction we go. The helpers you're exploring to shortcut to different states is one example. Another example could be autogenerated tracing (for debugging rather than testing). I thought about whether mocks would fit into this category, but I don't think so. The mocks we generate are so basic that we should expect the user to rewrite at least some of them, in which case we wouldn't want to clobber their edits by regenerating them.

So really I think of what we're generating here as just test scaffolding that's kept up to date.  It's still up to the user to take the scaffolding and do something meaningful with it. But whenever they're ready to do it, the scaffolding is ready and up-to-date for them.

Once you have that, you can layer on some nice to haves, like maybe a nice JS interface for users to be able to see and run their JS tests. This could easily be combined with the interactive simulator https://github.com/StateSmith/StateSmith/issues/267 to make it easy to debug and test your SM in one place.

Thinking about the levels you outlined, the levels in my mind were a little different.

- Level A: We have typed behaviors
  - all dependencies can be auto mocked in typed languages like C
  - JS has full working solution with simulation and test console

- Level B: No typed behaviors
  - user must manually mock behaviors in typed languages like C
  - JS still has full working solution with simulation and test console, though

I'm deliberately excluding options to dynamically keep the tests themselves up to date, I'm just focusing on template-based solutions where the examples rather than the tests are kept up to date.

adamfk, 2024-05-22:
This is a helpful discussion. I think I've been largely underestimating the need for tools to help new users get comfortable with StateSmith and hierarchical state machines. Especially when getting started with history vertices.

My particular needs are very different than that of new users. I more often need to exhaustively test a state machine controlling heavy mobile machinery like tractors. For that testing, I want integration tests (not unit tests). I added another ticket to track work towards this: https://github.com/StateSmith/StateSmith/issues/276

So maybe we can have two focus areas for making testing easier?
* Goal 1. Make it easy to learn StateSmith. Support simple verification.
* Goal 2. Make it easy to write integration tests for state machines.

# Goal 1 - learn and simple verification
From your Level A above:
> * all dependencies can be auto mocked
> * full working solution with simulation and test console

For this use case, I don't think we need to support multiple languages. We want to auto mock any external code anyway. I really like your idea of sticking to JS for this goal. It simplifies a lot.

Maybe initially we focus on an interactive simulation. This will probably help the most people. Initially, it will essentially mock all user action code [as shown here](https://github.com/StateSmith/StateSmith/issues/267#issuecomment-2123678289).

Once we have that working well, we can auto generate a JS test scaffold or have it be built into the interactive simulation. Users can then use JS to write basic unit tests.

Once that is working decently, then we have a choice of improving testing capability of Goal 1 (allowing user variables to be used) or moving towards Goal 2.

# Goal 2 - integration tests
This is where we generate language specific helpers and stuff like that.

<br>

Thoughts?

emmby, 2024-05-22:
Yes, totally agree! I think you're right and it makes sense that I've been more focused on the new user scenario and you've been more focused on the advanced user.

I like the focus of Goal 1. I also agree to focusing on JS, I'll pause the C work for now.