StateSmith/StateSmith #267: feature - generate an interactive web implementation by default

adamfk, 2024-05-08:
from: https://github.com/StateSmith/StateSmith/issues/264

> A web-implementation is a very convenient way for users, particularly new users, to interact with their state machines and verify they are working as expected. This can be a great way to test a new machine, or to test a change to an existing machine. They’re also great as interactive documentation for teams.

> https://statesmith.github.io/fundamentals-1/ provides many handy examples of interactive state machines, but the process to implement these is manual. Even once you get one generated, the difficulty of maintaining them is high due to the need to customize the config file for every new variable or method call for every language.

> With zero-config, StateSmith can easily generate these web implementations by default out of the box. Every MySM would generate an associated MySM.html and MySM.js file by default

This would be really nice.

There are a few hurdles I can think of off the top of my head:

1. it won't work very well with large state machines that have collapsed nested states. For those to simulate properly, we need to hook into draw.io. I took a look and it is possible for our plugin to control which state draw.io is in so that it could follow along an external simulation (or an actual live running fsm). donnerdachs and I started brainstorming ideas on this and then I got too busy for a while. Just messaged donnerdachs again. https://discord.com/channels/1056394875278462996/1062209027938996295/1145759599861973012
2. some state machines call external or global code. We can sometimes just ignore that, but sometimes it is necessary like calling `millis()` in an Arduino state machine.

adamfk, 2024-05-08:
The problem with collapsed states is that the SVG image being viewed in the browser needs large dynamic changes when you double click into a collapsed state. The image basically needs to be redrawn for the new contents. Maybe draw.io can already do that or could be made to do that?

If you haven't seen a collapsed state yet, check out this short video. About 1/3rd in, a collapsed state is entered. https://youtu.be/9czSDothuzM?si=jmGQm1VeTSWwyAPa

Collapsed states are incredibly useful with large diagrams. My largest state machine was around 300 states. It would have been a disaster without collapsing certain states.

We also don't necessarily need to support them for interactive web initially.

adamfk, 2024-05-16:
When mermaid is able to represent [parent states properly](https://github.com/StateSmith/StateSmith/issues/268) (or we could work around it for now), we could use it for the interactive web simulation.

This would solve 2 big problems:
* it works for any input diagram type
* it works with collapsed states

The beautiful thing about this is that it would work for any input diagram (plantuml, drawio xml file, yed, mermaid, ...) and not just drawio.svg files. It also wouldn't require the user to manually generate svg files.

It would also solve the collapsed state issue. We would just let mermaid render a fully expanded diagram. It might be a bit unwieldly for large diagrams, but I think we can provide some helpers.

For example, I asked chatgpt to create a 30 state plantuml state diagram with lots of interconnections and it [gave me this](https://www.plantuml.com/plantuml/duml/RTI_QiCm4CVn_Px2So5W_PUMagbRfuwbWqip1D8CZJELldjn9Dtzq7bA-4WvEFX6nzkw_AptxmkzFtrCXyFpzBgShs_hifw8Rjk7mFHBqzIlFSJGrBw8sMvMp7LpOhwR5mlTWbZi5iLIjoIMksMnqgs8cRcZcQlUFzioUC3N3Qi1jQmMrB4wKC_gGGDh08si4JIn9j3Cca4BQn6rCtMsmzO6Jk07XjM0MbOBwbWTg6Vre84rW4RM29fO4sXcpQ25jOWwMTlLvLxEr-No0FeZ6jiDB53B67DaWKR6kXCB93FcdbcWeB7_KWLpc3NvIWd8Q0gLCf3L52iL8ASfaVwlv6O6h3IpM6Cp1sQQ-PrvOMTUt9cNTkPbCDkiW3dIyNlIwGVIuKVIsIVIqMVIoPTc4NDjbdOi3wPJbxOttbvgekFf-hMzxly0).

One of the problems with really large diagrams like that is that it is often hard to follow transitions. In some cases, they don't fit on the screen. In the below image, I'm imagining that we could pop up helpers on transition mouse over.

[image]

This could actually solve another really large problem:
* we can't remotely communicate to our StateSmith plugin running inside the draw.io offline application (electron security setting)

It also wouldn't require running a local webserver so that the browser will allow dynamically modifying the draw.io generated [svg file](https://github.com/StateSmith/statesmith.github.io).

This would be awesome! If we get this working well, we could open it up so that a live running state machine could be viewed live with the simulation interface.

There's a lot of work here, but it would be a <u>**huge step**</u> forward for StateSmith.

adamfk, 2024-05-17:
https://js.cytoscape.org/ looks super promising of all the js based drawing tools that I've seen. It does auto layout and provides interactivity (mouse events, can dynamically animate/color). By default, you can pan/zoom, click edges to highlight. Supports popups.

* auto layout: https://ivis-at-bilkent.github.io/cytoscape.js-fcose/demo/demo-constraint.html
* animation (good for showing transition path): https://js.cytoscape.org/demos/animated-bfs/
* if we want to show hierarchy: https://js.cytoscape.org/demos/compound-nodes/
* fancy: https://pathwaycommons.github.io/cytoscape-sbgn-stylesheet/
* click nodes to see pop-up: https://js.cytoscape.org/demos/colajs-graph/

adamfk, 2024-05-22:
Basic working proof of concept (lacking diagram support): https://youtu.be/HLTxjsNtjfM

https://github.com/StateSmith/exp-test-gen/tree/6b4fe3a9d69ef599280a7af400d9cec06a7f7504/src2

@emmby are you able to explore whether mermaid diagrams are able to be dynamically updated like in https://statesmith.github.io/fundamentals-1/ ? That would be ideal. We have a lot of control with cytoscape, but we would also have to do a fair amount more work. mermaid is preferable.

Even while we work on diagram support, it might be useful to start to build this into StateSmith.

Thoughts?

emmby, 2024-05-24:
prototype: https://github.com/StateSmith/StateSmith-examples/pull/4

adamfk, 2024-05-25:
@emmby I made some changes to your prototype to support pseduo states and show an alternate approach to the visitor pattern. I also added a larger example with more element types.

Input:
[image]

Output:
```
stateDiagram
classDef active fill:yellow,stroke-width:2px;
state ORDER_MENU {
        state BEVERAGE {
                [*] --> BEVERAGE.(InitialState)
                state "$initial_state" as BEVERAGE.(InitialState)
                WATER
                WATER : WATER
                COFFEE
                COFFEE : COFFEE
                COFFEE : enter / #123; do_x()#59; #125;
                TEA
                TEA : TEA
        }
        state FOOD {
                [*] --> FOOD.(InitialState)
                state "$initial_state" as FOOD.(InitialState)
                JUNK
                JUNK : JUNK
                POTATO
                POTATO : POTATO
                SUSHI
                SUSHI : SUSHI
        }
        state "$entry_pt.to_history" as ORDER_MENU.(EntryPoint)(to_history)
        state "$exit_pt.1" as ORDER_MENU.(ExitPoint)(1)
        state "$H" as ORDER_MENU.(History)
        state ORDER_MENU.(ChoicePoint)(bev_choice) <<choice>>
}
WAITING
WAITING : WAITING
[*] --> ROOT.(InitialState)
state "$initial_state" as ROOT.(InitialState)
ORDER_MENU --> WAITING : EV3
BEVERAGE --> FOOD : EV2
BEVERAGE.(InitialState) --> WATER : 
WATER --> COFFEE : EV1
COFFEE --> TEA : EV1
TEA --> COFFEE : EV1
TEA --> BEVERAGE.(InitialState) : EV2
FOOD --> ORDER_MENU.(ChoicePoint)(bev_choice) : EV2
FOOD --> ORDER_MENU.(ExitPoint)(1) : ESCAPE
FOOD.(InitialState) --> JUNK : 
JUNK --> POTATO : EV1
POTATO --> SUSHI : EV1
SUSHI --> JUNK : EV1
ORDER_MENU.(EntryPoint)(to_history) --> ORDER_MENU.(History) : 
ORDER_MENU.(ExitPoint)(1) --> WAITING : 
ORDER_MENU.(ChoicePoint)(bev_choice) --> TEA : [a > b]
ORDER_MENU.(ChoicePoint)(bev_choice) --> BEVERAGE : else
WAITING --> ORDER_MENU.(EntryPoint)(to_history) : EV3
ROOT.(InitialState) --> WAITING : 
```

[image]

It actually works!!! Really impressive start :)

Changes:
* https://github.com/StateSmith/exp-test-gen/blob/adam-simple-sim/src3/LightSm.csx#L23-L25
* https://github.com/StateSmith/exp-test-gen/blob/adam-simple-sim/src3/LightSm.csx#L23-L25

emmby, 2024-05-26:
> @emmby I made some changes to your prototype to support pseduo states and show an alternate approach to the visitor pattern. I also added a larger example with more element types.
> * [...]

Quick question: both of these links look identical, is there a second link I should be looking at or just the one?

adamfk, 2024-05-26:
Oops. I think they were both supposed to be links in the same file.

adamfk, 2024-05-27:
@emmby I added highlighting of transitions to your work. All transitions taken during the current event dispatch are colored red. Previously taken transitions are colored green. Untaken transitions are left black. We can customize however we like :)

At some point, it would be interesting to try and visually indicate with color the history of edge traversal. Maybe enough to show the last few transition paths? Might end up being too noisy, but might also be helpful.

[image]

Commit: https://github.com/StateSmith/StateSmith-examples/commit/e07e146a5e75cfdb87041d3bd9c4b30b93595c18

adamfk, 2024-05-29:
- [ ] proof of concept

adamfk, 2024-05-30:
@emmby I created new tickets for simulator specific work.

These ones are up for grabs:
* #287 
* #288 
* #289 
* #290 

I've self assigned these to myself:
* #291 
* #292

emmby, 2024-06-07:
Any objection to marking this ticket closed?

adamfk, 2024-06-07:
@emmby I agree. Let's close it.

closed 2024-06-07 by emmby