StateSmith/StateSmith #280: One-click generation of code from models

emmby, 2024-05-25:
As a user, I would like to be able to generate code from a model in a single step. For example, if I have a model `Model.drawio.svg`, I would like to be able to (exact syntax TBD):

- `ss.cli Model.drawio.svg` and get the JS output, including JS state machine, html simulator, and test examples.
- `ss.cli --watch Model.drawio.svg` to enable watch mode, so that all the output files are regenerated whenever the source is changed.
- `ss.cli --watch --lang=c++,js Model.drawio.svg` to automatically generate C++ and JS code whenever the source is changed.
- If the model file doesn't exist, the CLI should prompt me if I want it to generate an example model before generating the target code. eg. `ss.cli NewModel.drawio.svg` would generate NewModel.drawio.svg, JS state machine, html simulator, etc., and `ss.cli --lang=C++,js NewModel.drawio.svg` would also generate C++ code.
- If I'm using the drawio plugin, I shouldn't even need to run the cli at all. The plugin should trigger all the codegen for me automatically on save.
- if I add `--config=foo.csx`, the codegen will use the specified config file for 

This is probably dependent on the "interim solution" in https://github.com/StateSmith/StateSmith/issues/264

adamfk, 2024-05-25:
Lots of good ideas in here! I'll start breaking some of them out into related tickets this weekend.

I've been wondering about a similar idea. Using the CLI without .CSX. right now the CLI can recursively find relevant CSX files and run them. We should be able to do the same with draw.io files although slower because the XML is sometimes base 64 encoded and compressed in the SVG. We have to decode and uncompress each SVG to see if it is a ss file.

Perhaps a user could use `fileName.ss.extension` or some other pattern to make it easier to auto find diagram files. Look for ".ss.".

Or they could use the ss.cli manifest or specify exactly the diagram as shown above. 

It would be really nice to have "ss.cli watch" work for multiple diagrams concurrently. Not needing a watch for every sm.

adamfk, 2024-05-26:
I forgot that the SVG will contain XML text with $STATEMACHINE : MySm. We can quickly and easily scan for that. No need to uncompress embedded diagram.