StateSmith/StateSmith #285: diagram only (no csx required)

adamfk, 2024-05-28:


adamfk, 2024-06-09:
@emmby you can try this out by installing a test release of the cli.

```
dotnet tool install --global StateSmith.Cli --version 0.8.2-diag-only-1
```

Then you can use the cli as normal, but it has another few options: (at bottom)
```
ss.cli run --help
Using settings directory: C:\Users\akruck\AppData\Roaming\StateSmith.Cli
StateSmith.Cli 0.8.2-diag-only-1+5fd14eca8da3bc97f3463498ccc03d61571f3b2b

  -h, --here         Runs code generation in this directory.

  -b, --rebuild      Ensures code generation is run. Ignores change detection.

  -u, --up           Searches upwards for manifest file.

  -r, --recursive    Recursive. Can't use with -i.

  -x, --exclude      Glob patterns to exclude

  -i, --include      Glob patterns to include. ex: `**/src/*.csx`. Can't use
                     with -r.

  --menu             Shows a terminal GUI with a choice menu. Don't use with
                     other options.

  --lang             Specifies programming language for transpiler. Ignored for
                     csx files.

  --no-sim-gen       Disables simulation .html file generation. Ignored for csx
                     files.
```

Use like `ss.cli run -hr --lang C99`. You should see something like this:

```
<snip>
── Checking diagram: `DiagOnlySm.plantuml` ───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Diagram settings: Lang: C99, NoSimGen: False
Change detection not implemented yet. Rebuild for diagram. Issue #328.
Running diagram: `DiagOnlySm.plantuml`

StateSmith Runner - Compiling file: `../../StateSmith.CliTest/Run/examples/1/DiagOnlySm.plantuml` (no state machine name specified).
StateSmith Runner - State machine `DiagOnlySm` selected.
StateSmith Runner - Writing to file `../../StateSmith.CliTest/Run/examples/1/DiagOnlySm.h`
StateSmith Runner - Writing to file `../../StateSmith.CliTest/Run/examples/1/DiagOnlySm.c`
StateSmith Runner - Writing to file `../../StateSmith.CliTest/Run/examples/1/DiagOnlySm.sim.html`
StateSmith Runner - Finished normally.

...Skipping diagram `yes1.plantuml` already run by csx file `yes.csx`.
<snip>
Finished running diagrams.
```

emmby, 2024-06-09:
Super cool! Seems to work well, it generated .c, .h, and .sim.html files for each of my diagrams

commit f57509c6 (2024-06-09): cli - `run` command now supports running diagrams with `--lang` option

closes #285
closed 2024-06-19 by adamfk