StateSmith/StateSmith #288: simulator - windows scroll bars

adamfk, 2024-05-30:
For some reason, there are a lot of scroll bars shown in the simulator when viewed with Chrome Windows.

[image]

emmby, 2024-05-30:
Can you try setting overflow: auto instead of scroll and see if that solves the problem? 
https://stackoverflow.com/questions/6689412/difference-between-html-overflow-auto-and-overflow-scroll

adamfk, 2024-05-30:
It got rid of one of the scroll bars.

[image]

Before:
[image]

After:
[image]

I can troubleshoot this one as I can reproduce easily. Firefox behaves the same way for me as well.

I think the main issue is that the content is too large.