StateSmith/StateSmith #295: simulator - change log ordering?

adamfk, 2024-05-31:
@emmby thoughts on changing the event log ordering to be normal (not reversed), but also auto scroll to show the latest? It would then function more like a traditional console output.

I've been playing around with the simulator and find it a bit hard to follow the log flow when reading upwards.

The changes are simple:

```diff
diff --git a/test-gen-2/src/LightSm.html b/test-gen-2/src/LightSm.html
index e00e740..48791e3 100644
--- a/test-gen-2/src/LightSm.html
+++ b/test-gen-2/src/LightSm.html
@@ -82,7 +82,7 @@
 
       .console tbody {
         display: flex;
-        flex-direction: column-reverse;
+        flex-direction: column;
         font-family: monospace;
       }
 
@@ -244,6 +244,8 @@ globalThis.light_yellow = ()=>{ addHistoryRow(new Date(), "Called mock light_yel
             row.appendChild(timeCell);
             row.appendChild(eventCell);
             document.querySelector('tbody').appendChild(row);
+            const logDiv = document.querySelector('div.history');
+            logDiv.scrollTop = logDiv.scrollHeight;
         }
 
         var sm = new LightSm();
```

emmby, 2024-05-31:
Yeah, I was looking at that too. The main problem with setting the scrollTop is that it changes your scroll position with every event change. So if you scroll somewhere in the window to look at something and an event happens in the background, it's going to jump you to that event. The advantage of the current reversed history is that you can scroll somewhere and you won't get jumped around if an event happens.

It feels like we should be able to get both (chronological order without jumping around) but I haven't figured out how yet.

adamfk, 2024-05-31:
Ah! Auto scrolling would be annoying when we have v2 simulations that are more "live".

What about an "auto scroll" check box like the Arduino terminal? Seems to work well.

PR #301 (2024-06-03): Show events in sidebar chronologically, and let user show/hide timestamps
- add a settings button that lets you enable/disable the timestamps
- order events chronologically instead of reverse-chronologically, closes #289
- non-mac devices should no longer show scrollbars (overflow: auto instead of scroll)

Closes #295
closed 2024-06-03 by emmby