StateSmith/StateSmith #302: Investigate why beadsorter SM is not highlighting states in simulator

emmby, 2024-06-02:


emmby, 2024-06-02:
The problem seems to be a mismatch in state names for child states.

StateSmith calls the `Start` state `SETUPCHECK_START`, but mermaid calls it `Start`. So when we we look for `document.querySelector('g[data-id=SETUPCHECK_START]')` we get no results

adamfk, 2024-06-03:
Good find. I believe I have a fix for this. Just need to move our first custom transformation pipeline step until after name conflict resolution occurs.

adamfk, 2024-06-03:
I think I fixed this in latest [push to main](https://github.com/StateSmith/StateSmith/commit/1cba4fba6207fc570fa551906748adce64c24097). Can you pull main and try again?

I unfortunately tagged the wrong ticket (303 instead of 302), but it worked out because 303 is an already closed PR. Working late problems 😛

emmby, 2024-06-03:
Hm, still doesn't seem to be working, I haven't gotten a chance to look into why yet. Here's a PR with the svg file: https://github.com/StateSmith/StateSmith/pull/307

adamfk, 2024-06-03:
Do you mind if I take this issue?

PR #311 (2024-06-04): 302 investigate why beadsorter sm is not highlighting states in simulator
fixes bead sorter simulation. Fixes #302 

keeps state capitalization
[image]

keeps event capitalization
[image]
closed 2024-06-04 by adamfk