StateSmith/StateSmith #308: Better highlighting support for large diagrams in simulator

emmby, 2024-06-03:
It's hard to see what's going on in large diagrams when states transition in the simulator.

Potential ideas:
- keep a zoomed out thumbnail when users zoom in: https://bumbu.me/svg-pan-zoom/demo/thumbnailViewer.html
- auto animate to the right state on transition if you're zoomed in. Would probably want to be able to turn this off.

adamfk, 2024-06-04:
Some svg animation libs that look promising:

Good for animating edge path:
- https://github.com/merri-ment/lazy-line-painter
    - good demo https://codepen.io/camoconnell/pen/wRYELj
  
General purpose svg animation. 
- http://snapsvg.io/
    - I believe I used this for my [high tech mini golf course](https://www.youtube.com/watch?v=TLmVZwPT6XU) (the animated elves that jump) 
    - Could also be used to add a bit of animation to sm vertex size (grow when entered, shrink to normal when exited)

emmby, 2024-06-04:
might be able to use `use`: https://stackoverflow.com/questions/25078634/can-i-re-use-the-same-svg-in-a-page-and-apply-different-css to include the svg twice (for thumbnail)

emmby, 2024-06-06:
Filed #320 for better edge highlighting support

PR #319 (2024-06-06): Scroll to the active state if it's offscreen
Closes #308
closed 2024-06-06 by adamfk