StateSmith/StateSmith #316: simulator - show `do` explicitly instead of implicitly?

adamfk, 2024-06-04:
When stepping through a simulation, I find it easier to show `do` events as explicit instead of implicit.

Implicit:
[image]

Explicit:
[image]

Thoughts @emmby?

emmby, 2024-06-06:
I'll take a look at https://github.com/StateSmith/StateSmith/discussions/194 and see. My instinct is to say that we shouldn't introduce something into the statemachine that the user didn't put there, but in this case it seems like that was a deliberate design decision, so trying to understand that better. (It does explain why my statemachine didn't automatically advance as expected when I ported it from my other tool, but at the time I thought it was related to how i didn't migrate my time handling code from the other tool)

adamfk, 2024-06-06:
I think showing `do` events explicitly may help users understand how their fsm functions. Even for myself, when I'm just clicking around in a simulation, I like seeing the `do` event so its easier to traverse the fsm easily.

Personally, I don't want to write `do` a bunch of times in my diagrams, but I like it shown in my simulation for clarity & because I didn't need to write it.

adamfk, 2024-06-06:
It's easy to experiment with as well. 1 liner: https://github.com/StateSmith/StateSmith/commit/85865d0197506a7c7b67bef014683fed16fa331a?diff=unified&w=0#diff-4ee324188f96729fb0a6f34fce582acb92dc5f49e77dc5d4f7dae4aa8eda463eR118

commit b9349025 (2024-06-21): sim - show implicit `do` edge trigger explicitly

close #316
closed 2024-06-21 by adamfk