StateSmith/StateSmith #317: Multiple history log messages

emmby, 2024-06-05:
It looks like there are multiple history logs when restoring a history state. Example below from https://github.com/StateSmith/StateSmith/blob/main/src/StateSmithTest/Output/Sim/diagrams/BeadSorter.sim.html has three entries for one dispatch:

```
Dispatched BUTTON0
➡️ Entered Running
🕑 History: transitioning to SortingMode__Dropoff.
➡️ Entered SortingMode
📝 History(Running) = SORTINGMODE
➡️ Entered SortingMode__Dropoff
⚡ FSM would execute action: picker.moveToAndShimmy(DEG_DROPOFF);
📝 History(Running) = SORTINGMODE__DROPOFF
```

Would it make sense to consolidate these down to a single entry?

adamfk, 2024-06-05:
For smaller designs, I think the current behavior is OK, but I could see having an optional feature to consolidate them being really nice in deeply nested diagrams. Probably want to enable it by default.

Reason for having ability to disable/enable: I like that the simulation largely shows what the FSM is doing. In this case, its highlighting that the history implementation is not yet optimized. Right now, it saves the history value for parent states even while it is transitioning to a child leaf state.

I'm actually finding the simulator a great tool for quickly identifying performance improvements. When I saw the all the guards being evaluated before we implemented https://github.com/StateSmith/StateSmith/issues/291, I was like 🤮 🤮 🤮  "I really need to replace that with a switch statement instead of a bunch of if statements!". It really helped highlight an area for optimization improvement. Just added https://github.com/StateSmith/StateSmith/issues/318 to track this.

Thoughts on how hard it would be to implement this feature?

emmby, 2024-06-06:
I think this is an example of a feature where the value is different for a statemachine author vs. a statesmith contributor.

I can see how it would be super useful to a contributor of statesmith to be able to see more verbose debugging events to be able to identify things like performance optimizations in statesmith.

For a user, I don't think these additional events are particularly useful. I found them a bit confusing. For me as the author of a statemachine, I want to know that history was restored and that's how I ended up in the state in the log message. The other messages were confusing for me.

Maybe it would make sense to have a verbose logging option for things that are more relevant to contributors? Or maybe just log them to the console?

adamfk, 2024-06-06:
I'm ok either way. Pros and cons to both, but not large in magnitude.

The main question for me is how to accomplish it. We don't currently have
the info built into StateSmith to easily support this.

Could post process printed logs to remove extra history events, but that
feels a bit hacky. Screen might occasionally show the glitch as extra DOM
rows are removed after being put on screen.

Could capture all logs before adding to screen. This would actually allow
some other future features like animations (if we want) or "stepping"
through an event dispatch.

Thoughts?

adamfk, 2024-06-06:
One other thing to consider is that some designs may have multiple history states in a diagram. There can also be multiple history states actively tracking at the same time (see [SANTAS_WORKSHOP](https://github.com/StateSmith/StateSmith/blob/da6ebad39666c840e18d7bd6396b416b670720b1/docs/history-vertex.md)).

[image]

Advanced history usage like this is rare, but whatever we do here shouldn't break it or make the simulator output be incorrect.

emmby, 2024-06-07:
Would we lose a lot by commenting out this line in SimWebGenerator?:

```
                // behavior.actionCode += $"this.tracer?.log('📝 History({historyVarName}) = {storedStateName}');";
```

In BeadSorter, the resulting event log is pretty clean and captures what I would expect it to capture, namely that history is taking us back to out last active substate (ReadEmptySlot__Sensor), and to do that we entered TrainingMode, ReadEmptySlot, and ReadEmptySlot__Sensor.

[image]

I haven't checked SANTAS_WORKSHOP yet but I expect it would be similar?

adamfk, 2024-06-07:
> Would we lose a lot by commenting out this line in SimWebGenerator?:
> 
> ```
>                 // behavior.actionCode += $"this.tracer?.log('📝 History({historyVarName}) = {storedStateName}');";
> ```
> 
> In BeadSorter, the resulting event log is pretty clean and captures what I would expect it to capture, namely that history is taking us back to our last active substate (ReadEmptySlot__Sensor), and to do that we entered TrainingMode, ReadEmptySlot, and ReadEmptySlot__Sensor.

Excellent point!

What if we add a `sim verbose flag` or something for all things like this? Disabled by default.

If people want to see all the gory details of what's really going on, they enable verbose mode **at generation time**.

PR #332 (2024-06-10): Disable detailed history logging for now.
Closes #317.
commit 1a70dd01 (2024-06-10): Disable detailed history logging for now.

Closes #317.
closed 2024-06-10 by adamfk