StateSmith/StateSmith #353: drawio - `RunnerSettings.dumpErrorsToFile` to true.

heywaj, 2024-07-08:
I have checked the wiki  and issues page , and i can't find  how/where to make  _RunnerSettings.dumpErrorsToFile` to true._ 
Thanks!
```
[build] StateSmith Runner - Compiling file: `../../Projects/xxxx/xxx/xxxx/SystemStateMachine.drawio` (no state machine name specified).
[build] Exception DrawIoException : failed getting attribute id from xml element.
[build] 
[build] You can enable exception detail dumping by setting `RunnerSettings.dumpErrorsToFile` to true.
[build] StateSmith Runner - Finished with failure.
```

adamfk, 2024-07-08:
Hello.

Documentation is still in progress (and needs to organized better), but I've recently added a bunch. Let me know if this helps: https://github.com/StateSmith/StateSmith/blob/main/docs/settings.md#smrunnersettings

Are you using a .csx file to run StateSmith?

adamfk, 2024-07-08:
FYI - I think I'll need to get you a new release to better diagnose this issue. I haven't encountered many draw.io parse failures, so the exception stack trace probably won't be enough for you to easily find it in the file.

When you get the stack trace info, can you please share it?

If you don't mind sharing your draw.io file, I can diagnose quickly and also use the relevant problematic snippet (not the whole thing) to create a new unit test. You can share it privately with me on discord if you like. Otherwise, it will probably be a few days before I have a release with better error reporting.

adamfk, 2024-07-08:
@heywaj I created a test build which should point out which XML node was problematic.

You should be able to simply update your csx file to use StateSmith `0.11.2-alpha-drawio-error-location-2` to get this error reporting. See https://github.com/StateSmith/StateSmith-examples/blob/main/drawio-xml-error-report/LightSm.csx#L2

You can also try it out in this example project: https://github.com/StateSmith/StateSmith-examples/tree/main/drawio-xml-error-report.

Hope this helps. Any info you can give me on how you came across a drawio file with this problem would be appreciated.

Thanks,

Adam

adamfk, 2024-07-16:
@heywaj I think we have this addressed now. I'm going to close, but we can re-open if you are still having issues.

closed 2024-07-16 by adamfk

heywaj, 2024-07-23:
@adamfk  
Thank you very much for your reply, I don't seem to get email notifications from the site so I didn't get back to you in time.
 I do using a csx file to run statesmith and I will try new test build. 
Once again thank you very much.