StateSmith/StateSmith #372: Support event payloads

chenlijun99, 2024-08-03:
Oftentimes events can carry additional payload data (e.g. new UART message received). It would be cool for it to be supported by StateSmith.

adamfk, 2024-08-03:
I agree. In the mean time, you can store the additional event information inside a state machine variable.

chenlijun99, 2024-08-03:
Thanks for the workaround

adamfk, 2024-08-03:
You are welcome :) I've used this work around a number of times successfully.

Even though it might take a while before we implement this feature, I'd be happy to hear ideas/thoughts on it.

renestraub, 2025-05-23:
I second this request completely. I use this pattern all the time when working with state machines. Using a SM variable is not thread safe when more than one SM and/or IRQs are involved. In the end you have to create (thread/IRQ) safe queue to hold the payload.