StateSmith/StateSmith #375: improve exception printing & wording around `propagate exceptions`

adamfk, 2024-08-08:
Before propagate exceptions enabled:
[image]

This looks weird like part of the Vertex info was cut off. I might have a flushing/buffer issue.

Running again:
[image]

I should also recommend users not use 'propagate exceptions' unless they want a detailed stack trace, but no context.

When propagate exceptions are enabled it is worse for user:
[image]

With dumping to file:
[image]

[image]

adamfk, 2024-08-08:
Related areas of code:

https://github.com/StateSmith/StateSmith/blob/556c4612fddd1fab378ce7287623f260ad6813b7/src/StateSmith/Runner/SmRunnerInternal.cs#L79-L103

We throw an exception always because some `dotnet-script` versions fail to set exit code properly without an exception: https://github.com/StateSmith/StateSmith/blob/556c4612fddd1fab378ce7287623f260ad6813b7/src/StateSmith/Runner/SmRunner.cs#L117-L118

Joebeazelman, 2024-08-08:
Instead of rethrowing using 'throw', try throw new Exception(e).

commit cc35991c (2024-08-12): improve exception printing & wording around propagate exceptions. closes #375
closed 2024-08-19 by adamfk