StateSmith/StateSmith #377: Support for Ada

Joebeazelman, 2024-08-09:
I would like to support Ada as a language.

adamfk, 2024-08-09:
I've heard many great things about Ada. This is awesome :)

I added a small repo to help with this initial exploration work: https://github.com/StateSmith/new-language-kickoff

Looking forward to it!

Joebeazelman, 2024-08-09:
Thanks. I appreciate the work you've put into this project and your enthusiasm. I am very impressed with the simulation demos and I think you're off to a great start. Unfortunately, as I dug deeper into the actual generated code, I'm not sure this is the right tool for my needs.

First, the code relies on nullable references. It's a major source of bugs and should be used sparingly, especially in something as critical as a state machine. 

Second, the generator produces a lot of unnecessary code. It generates a function for every state, making the code very difficult to follow. It also generates code for outputting states as strings, which is already built-in in both C# and JavaScript.

Finally, the entire state machine is far better represented using a transition table. It's more succinct, readable, and easier to maintain. I read somewhere you decided against it because you believe it's memory inefficient. This is a common misconception. Modern compilers do an excellent job of optimizing data structures, but it requires some expertise to fully exploit them.

I'm not sure whether this is a good fit for my needs.

adamfk, 2024-08-09:
If you'd like, you can write your own fully custom code generator. You can
hook it in without changing StateSmith if you are using a .csx file. Uses
dependency injection. Much of StateSmith can be customized from a CSX file.

Your code wouldn't need to transpile. You receive a state machine graph
from StateSmith and then the rest is up to you.

Then you get the benefit of diagram parsing and state machine graph
transformations.

There are settings to disable the "to string" functions if you want.

I'm personally not worried about null references in the generated code
because of the thorough specification test coverage. Maybe you could run
into an issue if you didn't respect concurrency.

There is no perfect Algorithm for every need. There are always trade offs.
The current algorithm works well for every need I've come across, but could
always be more optimized for specific needs/desires.

I'd be happy to support you if you wanted to write a transition table
algorithm if you like. It might not be too hard if the state machine is
flattened first (remove hierarchy).

The down side of larger tables is for architectures like AVR that can't
store data in flash without a bunch of PROGMEM work arounds.

Joebeazelman, 2024-08-10:
Sounds great! Fortunately, you made the generator very flexible. 

The real value for me is the plantuml parsing and the state machine simulation, which doesn't have to be generated in the target language. 

Presuming I go with the CSX approach, what are the next steps? Continue with hand translating the sample? 

----
On a side note, I strongly encourage you to explore some of the benefits of using modern C++ over C. Not for its OOP constructs, but for its optimizations. 

A state machine transition table can be optimized away to almost nothing in C++. If you think about it, the table or its values never change at runtime, giving the compiler enormous leverage. For instance, statements like **print transitions[OnState]**, where the table and its values are declared constants, are substituted with **print OffState** at compile time, eliminating the **transitions** table entirely.

adamfk, 2024-08-10:
Are you thinking of creating a new algorithm for StateSmith that uses a
table and can be transpiled to other languages?

Or a custom ada code generator for StateSmith? No transpiling.

Or something completely custom just for your use case?

Joebeazelman, 2024-08-10:
I'm planning  on doing the first one. Adding a new language with its own algorithm. I just completed it using lightsm. 

```
package LightSm is
    type State is (START, OFF, ON_GROUP, ON1, ON2);
    type Event is (DIM, INC, OFF);

    Current_State : State := START;

    type Transition_Table is array (State, Event) of State;

    Transitions : constant Transition_Table :=
       (START    => (DIM => OFF, INC => OFF, OFF => OFF),
        OFF      => (DIM => OFF, INC => ON1, OFF => OFF),
        ON_GROUP => (DIM => OFF, INC => ON2, OFF => OFF),
        ON1      => (DIM => OFF, INC => ON2, OFF => OFF),
        ON2      => (DIM => ON1, INC => ON2, OFF => OFF));

    function Next_State (Current_Event : Event) return State is
       (Transitions (Current_State, Current_Event));

end LightSm;

```

adamfk, 2024-08-10:
nice! I've been wanting to add less general purpose algorithms, but most of my time has been on ease of use improvements lately.

Because we are looking at a new algorithm that will be applied to all supported languages, can we start prototyping it in C? It would make things a lot easier for me.

We can still use the [new language kickoff](https://github.com/StateSmith/new-language-kickoff/blob/main/README.md#step-1---manual-code-translation) repo, but focus on the algorithm and use C first.

I'm keen to see your example in more detail. I don't quite understand how the transition table will work with guards, actions, entry/exit and other features. It looks like it maybe requires a transition every event? Let's make something small that can be run with the same [`test-harness.c`](https://github.com/StateSmith/new-language-kickoff/blob/main/c/test-harness.c#L17-L23).

Have you considered a constant table of function pointers. It could have "nop()" like functions instead of nulls for when a state doesn't care about an event.

| state         | event: off        | event: dim        | event: inc        |
| ------------ | ----------------- | ----------------- | ----------------- |
| OFF          | unused_event()    | unused_event()    | off_inc_handler() |
| ON1          | on1_off_handler() | on1_dim_handler() | on1_inc_handler() |
| ON2          | …                 | …                 | …                 |

Joebeazelman, 2024-08-10:
Awesome! 

The problem I see with many code generators is the kitchen sink design approach where a lot of complex and overly generalized code is generated to handle developer use case. It almost always backfires, often resulting in inefficient, inflexible, and hard to comprehend. It’s precisely the reason why some of the most popular developer tools tend to be simple, hence the popularity of StarUML and PlantUML.

Consequently, the generated code should be as simple as possible. A state machine is fundamentally tracks state, taking an event as an input and returning the next state as an output. How it’s used should be left to the developer, however, how it is implemented should be left to the generator.

The best approach generate the most common implementation schemes with some configuration specified in the diagram, or in a separate file. Where this tool adds value is constructing the transitions in code, which is tedious and error-prone to do manually. 

Consequently, I see three implementation levels: 

- Minimum level: A state machine with a transition table of state and event values. Substates are flattened. Client writes their own dispatcher function. Roughly the code I submitted.
- Medium level: A state machine with a generated dispatcher function.
- Maximum: A medium level state machine using a transition table of function pointers. 

You can isolate the dispatch function in a separate file so the transition tables can be regenerated whenever the diagram is modified with perhaps a warning if it requires modification to dispatch function. You could refer the client to samples of how each one can be used.

The guards, actions, entry and exit functions could be stubbed in as comments, but only if they are specified in the actual diagram, otherwise it adds excessive noise to the code.  

How do you think?  What's my next step from here? I am proficient in C#, C, and a little JavaScript. I'll follow up with a maximum level sample using Ada.

adamfk, 2024-08-11:
One of my goals with StateSmith is to provide a tool to make it easier to
code generate state machines. We don't need to all reinvent the wheel from
scratch.

Your suggestion is fundamentally different than where StateSmith is
currently heading, but I think that's cool. It gives people more options
which is always good. We just need to document the different algorithms to
help users. Because the approach is fundamentally different and likely
won't have the same feature set, I plan to help support you as I'm able,
but you'd be the primary driver of the new algorithm. Sound good?

I think a helpful next step is to manually implement the state machine that
satisfies the c test harness linked above.

In a couple days, I'll put together an example for how you can hook into
StateSmith to add a new algorithm.

You can also explore some of the more advanced examples in the StateSmith
examples repo.

adamfk, 2024-08-12:
This example is particularly relevant: https://github.com/StateSmith/StateSmith-examples/tree/main/visitor-graph-1

It doesn't show how to hook in custom code generation, but it does show how you can get and "visit" the state machine graph.

Also, if you are new to debugging C# scripts (.csx), check out the resources here: https://github.com/StateSmith/StateSmith/wiki/csx

adamfk, 2024-09-19:
@Joebeazelman how's it going?

I recently added support for a new algorithm `Balanced2` (the new default) that uses switch statements instead of function pointers. I know you mentioned that nullable function pointers are a concern for you.

I also added support for Java and Python. Java was relatively easy because it is really close to C#. Python was more difficult. For both of them, I decided to only support the `Balanced2` algorithm for now.

Anyway, it reminded me that this work can be pretty challenging and thought I would see how it was going.

Joebeazelman, 2024-09-19:
I've decided to go with a manual approach. It's really not worth the time and effort to customize a tool for my purposes. I was hoping I could find something I could easily modify and get running.

adamfk, 2024-09-19:
Makes sense. The work is non-trivial. I'm going to close the issue for now. We can re-open if there is interest in the future.

closed 2024-09-19 by adamfk